project-chip/connectedhomeip #27188: [BUG] Wrong formula for Immuniance

s-hadinger, 2023-06-11:
### Reproduction steps

I suspect a typo or misunderstanding about Zigbee and Matter Illuminance measurement, due to the absence of parenthesis.

The spec says:
`measured_value = 10000 * log10 lux + 1`

Which was interpreted as
`measured_value = 10000 * log10(lux) + 1`

When it actually meant:
`measured_value = 10000 * log10(lux + 1)`

The current formula is able to express very precise values just above 1 lux but none below 1 lux, which seems irrelevant:

lux value|Illuminance value
:---|:---
0|0 (although it may be interpreted as invalid)
0.5|_invalid_
0.9|_invalid_
0.99|_invalid_
1.0|1
1.0002|2
1.0004|3
1.1|414
2|3011
10|10001
100|20001
500|26990

While when using `10000 * log10(lux + 1)` it gives full range from low to high illuminance values

lux value|Illuminance value
:---|:---
0|1
0.00002|2
0.0004|3
0.5|1760
0.9|2787
0.99|2988
1.0|3010
1.1|3222
2|4771
10|10413
100|20043
500|26999

I propose to clarify and fix the formula. For controllers using the old formula, the result is off by 1, an external application can easily fix the value by just substracting `1`.

### Bug prevalence

Always, bug in specification.

### GitHub hash of the SDK that was being used

8f66f4215bc0708efc8cc73bda80620e67d8955f

### Platform

other

closed 2023-06-12 by bzbarsky-apple

s-hadinger, 2023-06-12:
Thanks, I understand that changing this would create some inaccuracies at low illuminance.

Where is it possible to open a spec issue?

bzbarsky-apple, 2023-06-12:
https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/ but you probably need to be a CSA member to do that...

s-hadinger, 2023-06-12:
Thanks but we will not be able to do it.

leorozendaal, 2023-06-14:
Long time ago, I proposed https://groups.csa-iot.org/wg/matter-tsg/document/18652 (Matter WG members access only) to be able to also represenent measured values < 1 lux