project-chip/connectedhomeip #27190: [BUG] AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout

ngovancuong2735, 2023-06-12:
### Reproduction steps

I copied the chip-tool build file from my PC to the mini PC.
PC: ubuntu 20.04.06, use wifi/eth
mini PC: ubuntu 20.04.06, only eth, use USB-BLE (4.0).

When I test chip-tool on the mini PC I get the error `Timeout waiting for mDNS resolution`. And lead to the error `AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout`.
I see the difference between PC and mini PC in the 2 log snippets below:
PC:
```
[1686534322.709825][15051:15053] CHIP:DMG: InvokeResponseMessage =
[1686534322.709841][15051:15053] CHIP:DMG: {
[1686534322.709856][15051:15053] CHIP:DMG: 	suppressResponse = false, 
[1686534322.709871][15051:15053] CHIP:DMG: 	InvokeResponseIBs =
[1686534322.709894][15051:15053] CHIP:DMG: 	[
[1686534322.709909][15051:15053] CHIP:DMG: 		InvokeResponseIB =
[1686534322.709932][15051:15053] CHIP:DMG: 		{
[1686534322.709948][15051:15053] CHIP:DMG: 			CommandDataIB =
[1686534322.709965][15051:15053] CHIP:DMG: 			{
[1686534322.709980][15051:15053] CHIP:DMG: 				CommandPathIB =
[1686534322.709998][15051:15053] CHIP:DMG: 				{
[1686534322.710016][15051:15053] CHIP:DMG: 					EndpointId = 0x0,
[1686534322.710034][15051:15053] CHIP:DMG: 					ClusterId = 0x31,
[1686534322.710052][15051:15053] CHIP:DMG: 					CommandId = 0x7,
[1686534322.710069][15051:15053] CHIP:DMG: 				},
[1686534322.710087][15051:15053] CHIP:DMG: 				
[1686534322.710104][15051:15053] CHIP:DMG: 				CommandFields = 
[1686534322.710121][15051:15053] CHIP:DMG: 				{
[1686534322.710140][15051:15053] CHIP:DMG: 					0x0 = 0, 
[1686534322.710158][15051:15053] CHIP:DMG: 					0x2 = NULL
[1686534322.710175][15051:15053] CHIP:DMG: 				},
[1686534322.710191][15051:15053] CHIP:DMG: 			},
[1686534322.710212][15051:15053] CHIP:DMG: 			
[1686534322.710227][15051:15053] CHIP:DMG: 		},
[1686534322.710249][15051:15053] CHIP:DMG: 		
[1686534322.710264][15051:15053] CHIP:DMG: 	],
[1686534322.710290][15051:15053] CHIP:DMG: 	
[1686534322.710304][15051:15053] CHIP:DMG: 	InteractionModelRevision = 1
[1686534322.710318][15051:15053] CHIP:DMG: },
[1686534322.710371][15051:15053] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0007
[1686534322.710398][15051:15053] CHIP:CTL: Received ConnectNetwork response, networkingStatus=0
[1686534322.710420][15051:15053] CHIP:CTL: Successfully finished commissioning step 'WiFiNetworkEnable'
[1686534322.710437][15051:15053] CHIP:CTL: Commissioning stage next step: 'WiFiNetworkEnable' -> 'FindOperational'
[1686534322.710455][15051:15053] CHIP:CTL: Performing next commissioning step 'FindOperational'
[1686534322.710470][15051:15053] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000001]
[1686534322.710486][15051:15053] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1686534322.710510][15051:15053] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 1 --> 2
[1686534322.710909][15051:15053] CHIP:DMG: ICR moving to [AwaitingDe]
[1686534322.911514][15051:15053] CHIP:DIS: Checking node lookup status after 201 ms
**[1686534323.735763][15051:15053] CHIP:DIS: Lookup clearing interface for non LL address
[1686534323.735855][15051:15053] CHIP:DIS: UDP:10.10.50.29%wlp7s0:5540: new best score: 2**
[1686534323.735880][15051:15053] CHIP:DIS: Checking node lookup status after 1025 ms
[1686534323.735940][15051:15053] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:10.10.50.29:5540 while in state 2
[1686534323.735972][15051:15053] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 2 --> 3
[1686534323.736070][15051:15053] CHIP:IN: SecureSession[0x7fb2e0059360]: Allocated Type:2 LSID:41396
[1686534323.736109][15051:15053] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000000001
[1686534323.737218][15051:15053] CHIP:ATM: CHIP_ERROR SendMessage
[1686534323.737281][15051:15053] CHIP:EM: <<< [E:33849i S:0 M:55156334] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[1686534323.737331][15051:15053] CHIP:IN: (U) Sending msg 55156334 to IP address 'UDP:10.10.50.29:5540'
[1686534323.737488][15051:15053] CHIP:SC: Sent Sigma1 msg
[1686534323.737520][15051:15053] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 3 --> 4
[1686534323.924013][15051:15053] CHIP:ATM: CHIP_ERROR TransportMgrBase::HandleMessageReceived

```
Mini PC:
```
[1686536003.257343][1965:1967] CHIP:EM: Found matching exchange: 65376i, Delegate: 0x7f3ea003e808
[1686536003.257382][1965:1967] CHIP:DMG: ICR moving to [ResponseRe]
[1686536003.257436][1965:1967] CHIP:DMG: InvokeResponseMessage =
[1686536003.257461][1965:1967] CHIP:DMG: {
[... 62 lines cut ...]
[1686536047.954227][1965:1967] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1686536048.260018][1965:1967] CHIP:DIS: Checking node lookup status after 45001 ms
[1686536048.260084][1965:1967] CHIP:DIS: OperationalSessionSetup[1:0000000000000037]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
[1686536048.260109][1965:1967] CHIP:DIS: Retrying operational DNS-SD discovery. Attempts remaining: 2
[1686536048.260546][1965:1967] CHIP:CTL: Session establishment failed for <0000000000000037, 1>, error: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout.  Next retry expected to get a response to Sigma1 or fail within 60 seconds
```

PC has but mini PC doesn't:

**[1686534323.735763][15051:15053] CHIP:DIS: Lookup clearing interface for non LL address
[1686534323.735855][15051:15053] CHIP:DIS: UDP:10.10.50.29%wlp7s0:5540: new best score: 2**

This error is because I copied the chip-tool file the wrong way or because my mini PC lacks some services. Or is it coming from my USB-BLE?

### Bug prevalence

every

### GitHub hash of the SDK that was being used

master

### Platform

other

bzbarsky-apple, 2023-06-12:
The error indicates DNS-SD resolution failed.  Are you compiling with minimal mdns, or using avahi?  In the latter case, what happens when you try to manually run the relevant avahi commands?

ngovancuong2735, 2023-06-12:
I check both of these services and they are still working normally.
```
cuong@cuongMini:~/matter/out$ sudo systemctl status mdns.service 
● mdns.service
     Loaded: loaded (/etc/init.d/mdns; generated)
     Active: active (running) since Mon 2023-06-12 13:46:39 +07; 5min ago
       Docs: man:systemd-sysv-generator(8)
    Process: 783 ExecStart=/etc/init.d/mdns start (code=exited, status=0/SUCCESS)
      Tasks: 1 (limit: 4482)
     Memory: 1.3M
     CGroup: /system.slice/mdns.service
             └─791 /usr/sbin/mdnsd

Thg 6 12 13:47:15 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: ProbeCount 0; restarting probing after 1024-tick pause due to possibly spurious multicast conflict (1/1) via interface 101169790>
Thg 6 12 13:47:16 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: Received from 10.10.60.55:5353   16 cuongMini.local. AAAA fd11:1111:1122:2222:b5c8:f63a:a8ae:a6a9
Thg 6 12 13:47:16 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: ProbeCount 2; restarting probing after 1024-tick pause due to possibly spurious multicast conflict (1/1) via interface 101169790>
Thg 6 12 13:47:17 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: Received from fd11:1111:1122:2222:b5c8:f63a:a8ae:a6a9:5353   16 cuongMini.local. AAAA fd11:1111:1122:2222:b5c8:f63a:a8ae:a6a9
Thg 6 12 13:47:17 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: ProbeCount 0; will deregister   16 cuongMini.local. AAAA fe80::a290:aa28:30d6:fd67 due to multicast conflict via interface 10116>
Thg 6 12 13:47:17 cuongMini mDNSResponder[791]: Default: Local Hostname cuongMini.local already in use; will try cuongMini-2.local instead
Thg 6 12 13:47:18 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: Received from 10.10.60.55:5353   17 55.60.10.10.in-addr.arpa. PTR cuongMini.local.
Thg 6 12 13:47:18 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: Unexpected conflict discarding   19 55.60.10.10.in-addr.arpa. PTR cuongMini-2.local.
Thg 6 12 13:47:18 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: Received from 10.10.60.55:5353   17 7.6.d.f.6.d.0.3.8.2.a.a.0.9.2.a.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. PTR cuongMini.loca>
Thg 6 12 13:47:18 cuongMini mDNSResponder[791]: Default: mDNSCoreReceiveResponse: Unexpected conflict discarding   19 7.6.D.F.6.D.0.3.8.2.A.A.0.9.2.A.0.0.0.0.0.0.0.0.0.0.0.0.0.8.E.F.ip6.arpa. PTR cuongMini-2.lo>
```
```
cuong@cuongMini:~/matter/out$ sudo systemctl status avahi-daemon.service 
● avahi-daemon.service - Avahi mDNS/DNS-SD Stack
     Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-06-12 13:46:39 +07; 5min ago
TriggeredBy: ● avahi-daemon.socket
   Main PID: 570 (avahi-daemon)
     Status: "avahi-daemon 0.7 starting up."
      Tasks: 2 (limit: 4482)
     Memory: 1.3M
     CGroup: /system.slice/avahi-daemon.service
             ├─570 avahi-daemon: running [cuongMini.local]
             └─626 avahi-daemon: chroot helper

Thg 6 12 13:46:42 cuongMini avahi-daemon[570]: Joining mDNS multicast group on interface enp1s0.IPv6 with address fe80::a290:aa28:30d6:fd67.
Thg 6 12 13:46:42 cuongMini avahi-daemon[570]: New relevant interface enp1s0.IPv6 for mDNS.
Thg 6 12 13:46:42 cuongMini avahi-daemon[570]: Registering new address record for fe80::a290:aa28:30d6:fd67 on enp1s0.*.
Thg 6 12 13:46:42 cuongMini avahi-daemon[570]: Joining mDNS multicast group on interface enp1s0.IPv4 with address 10.10.60.55.
Thg 6 12 13:46:42 cuongMini avahi-daemon[570]: New relevant interface enp1s0.IPv4 for mDNS.
Thg 6 12 13:46:42 cuongMini avahi-daemon[570]: Registering new address record for 10.10.60.55 on enp1s0.IPv4.
Thg 6 12 13:47:14 cuongMini avahi-daemon[570]: Leaving mDNS multicast group on interface enp1s0.IPv6 with address fe80::a290:aa28:30d6:fd67.
Thg 6 12 13:47:14 cuongMini avahi-daemon[570]: Joining mDNS multicast group on interface enp1s0.IPv6 with address fd11:1111:1122:2222:b5c8:f63a:a8ae:a6a9.
Thg 6 12 13:47:14 cuongMini avahi-daemon[570]: Registering new address record for fd11:1111:1122:2222:b5c8:f63a:a8ae:a6a9 on enp1s0.*.
Thg 6 12 13:47:24 cuongMini avahi-daemon[570]: Withdrawing address record for fe80::a290:aa28:30d6:fd67 on enp1s0.
```
```
cuong@cuongMini:~/matter/out$ avahi-browse -a
+ enp1s0 IPv6 9c:65:f9:3e:ea:00                             _lumismarthome._tcp  local
+ enp1s0 IPv4 9c:65:f9:3e:ea:00                             _lumismarthome._tcp  local
+ enp1s0 IPv6 HIKVISION LM-AC270412-BU2/SL - D40861183      _CGI._tcp            local
+ enp1s0 IPv4 HIKVISION LM-AC270412-BU2/SL - D40861183      _CGI._tcp            local
+ enp1s0 IPv6 HIKVISION LM-AC270412-BU2/SL - D40861183      _psia._tcp           local
+ enp1s0 IPv4 HIKVISION LM-AC270412-BU2/SL - D40861183      _psia._tcp           local
+ enp1s0 IPv6 HIKVISION LM-AC270412-BU2/SL - D40861183      Web Site             local
+ enp1s0 IPv4 HIKVISION LM-AC270412-BU2/SL - D40861183      Web Site             local
+ enp1s0 IPv6 OpenThread BorderRouter #5BA7                 _meshcop._udp        local
+ enp1s0 IPv4 OpenThread BorderRouter #5BA7                 _meshcop._udp        local
+ lo IPv4 OpenThread BorderRouter #5BA7                 _meshcop._udp        local
```

bzbarsky-apple, 2023-06-12:
The real question is whether `avahi-browse -a` shows the operational advertisement that then fails to resolve.

ngovancuong2735, 2023-06-13:
In my PC:
```
+ wlp7s0 IPv4 265D4094BAA0FE09                              _matterc._udp        local
+ wlp7s0 IPv6 265D4094BAA0FE09                              _matterc._udp        local

```
In my Mini PC:
cuong@cuongMini:~$ avahi-browse -a
```
+ enp1s0 IPv6 9c:65:f9:3e:ea:00                             _lumismarthome._tcp  local
+ enp1s0 IPv4 9c:65:f9:3e:ea:00                             _lumismarthome._tcp  local
+ enp1s0 IPv6 HIKVISION LM-AC270412-BU2/SL - D40861183      _CGI._tcp            local
+ enp1s0 IPv4 HIKVISION LM-AC270412-BU2/SL - D40861183      _CGI._tcp            local
+ enp1s0 IPv6 HIKVISION LM-AC270412-BU2/SL - D40861183      _psia._tcp           local
+ enp1s0 IPv4 HIKVISION LM-AC270412-BU2/SL - D40861183      _psia._tcp           local
+ enp1s0 IPv6 HIKVISION LM-AC270412-BU2/SL - D40861183      Web Site             local
+ enp1s0 IPv4 HIKVISION LM-AC270412-BU2/SL - D40861183      Web Site             local
+ enp1s0 IPv6 OpenThread BorderRouter #5BA7                 _meshcop._udp        local
+ enp1s0 IPv4 OpenThread BorderRouter #5BA7                 _meshcop._udp        local
+ enp1s0 IPv6 OpenThread BorderRouter #751F                 _meshcop._udp        local
+ enp1s0 IPv4 OpenThread BorderRouter #751F                 _meshcop._udp        local
- enp1s0 IPv6 OpenThread BorderRouter #5BA7                 _meshcop._udp        local
- enp1s0 IPv4 OpenThread BorderRouter #5BA7                 _meshcop._udp        local
```

bzbarsky-apple, 2023-06-13:
@ngovancuong2735 None of that is showing Matter _operational_ advertisements (which would be of type `_matter._tcp`).

jacqueshsu, 2023-11-03:
@ngocuong86
Did you fix your issue? I got the same problem as your PC.

igor90ns, 2023-12-28:
Hi, is there any news regarding this error? I am having same issue.
My setup is Chiptool app and ESP32 with latest fw from zerocode
When I try to pair device it always fail to find Operational device (but i can see it in router dhcp that is successfully connected)

Logs from chip tool
`

2023-12-28 08:56:45.400  3717-3717  CTL                     com.google.chip.chiptool             D  JNI_OnLoad() called
2023-12-28 08:56:45.400  3717-3717  CTL                     com.google.chip.chiptool             D  Loading Java class references.
2023-12-28 08:56:45.400  3717-3717  CTL                     com.google.chip.chiptool             D  Java class references loaded.
2023-12-28 08:56:45.400  3717-3717  DL                      com.google.chip.chiptool             D  AndroidChipPlatform JNI_OnLoad() called
2023-12-28 08:56:45.400  3717-3717  SPT                     com.google.chip.chiptool             E  JavaVM is already set
2023-12-28 08:56:45.400  3717-3717  DL                      com.google.chip.chiptool             D  Loading Java class references.
2023-12-28 08:56:45.400  3717-3717  DL                      com.google.chip.chiptool             D  Java class references loaded.
2023-12-28 08:56:45.400  3717-3717  DL                      com.google.chip.chiptool             D  BleConnectCallbackJNI_OnLoad
2023-12-28 08:56:45.400  3717-3717  DL                      com.google.chip.chiptool             D  Initializing tracing for android
2023-12-28 08:56:45.404  3717-3717  Compatibil...geReporter com.google.chip.chiptool             D  Compat change id reported: 263076149; UID 10309; state: DISABLED
2023-12-28 08:56:45.409  3717-3717  BluetoothAdapter        com.google.chip.chiptool             I  BluetoothAdapter() : com.google.chip.chiptool
2023-12-28 08:56:45.418  3717-3717  Compatibil...geReporter com.google.chip.chiptool             D  Compat change id reported: 270306772; UID 10309; state: DISABLED
2023-12-28 08:56:45.420  3717-3717  Compatibil...geReporter com.google.chip.chiptool             D  Compat change id reported: 235355681; UID 10309; state: ENABLED
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  newDeviceController() called
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  Setting attestation nonce to random value
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  Setting CSR nonce to random value
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  Setting attestation nonce to random value
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  Setting CSR nonce to random value
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  Setting attestation nonce from parameters
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  Setting CSR nonce from parameters
2023-12-28 08:56:45.426  3717-3717  TOO                     com.google.chip.chiptool             D  KVS: Getting key AndroidDeviceControllerKey
2023-12-28 08:56:45.426  3717-3717  CTL                     com.google.chip.chiptool             D  No existing credentials provided: generating ephemeral local NOC chain with OperationalCredentialsIssuer
2023-12-28 08:56:45.431  3717-3717  TOO                     com.google.chip.chiptool             D  KVS: Getting key AndroidCARootCert1
2023-12-28 08:56:45.431  3717-3717  CTL                     com.google.chip.chiptool             D  Generating NOC
2023-12-28 08:56:45.437  3717-3717  IN                      com.google.chip.chiptool             D  UDP::Init bind&listen port=5541
2023-12-28 08:56:45.437  3717-3717  IN                      com.google.chip.chiptool             D  UDP::Init bound to port=5541
2023-12-28 08:56:45.437  3717-3717  IN                      com.google.chip.chiptool             D  UDP::Init bind&listen port=5541
2023-12-28 08:56:45.437  3717-3717  IN                      com.google.chip.chiptool             D  UDP::Init bound to port=5541
2023-12-28 08:56:45.437  3717-3717  IN                      com.google.chip.chiptool             D  BLEBase::Init - setting/overriding transport
2023-12-28 08:56:45.437  3717-3717  IN                      com.google.chip.chiptool             D  TransportMgr initialized
2023-12-28 08:56:45.437  3717-3717  FP                      com.google.chip.chiptool             D  Initializing FabricTable from persistent storage
2023-12-28 08:56:45.437  3717-3717  TOO                     com.google.chip.chiptool             D  KVS: Getting key g/lkgt
2023-12-28 08:56:45.437  3717-3717  TS                      com.google.chip.chiptool             D  Last Known Good Time: 2023-09-04T15:59:41
2023-12-28 08:56:45.437  3717-3717  TOO                     com.google.chip.chiptool             D  KVS: Getting key g/fidx
2023-12-28 08:56:45.437  3717-3717  TOO                     com.google.chip.chiptool             D  KVS: Getting key f/1/n
2023-12-28 08:56:45.437  3717-3717  TOO                     com.google.chip.chiptool             D  KVS: Getting key f/1/r
2023-12-28 08:56:45.437  3717-3717  TOO                     com.google.chip.chiptool             D  KVS: Getting key f/1/m
[... cut ...]
I (124643) NimBLE: GATT procedure initiated: indicate;
I (124643) NimBLE: att_handle=14

I (124723) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (127243) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (129743) NimBLE: GATT procedure initiated: indicate;
I (129743) NimBLE: att_handle=14

I (129823) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (132343) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12`

suktry, 2024-01-09:
Hello, I encountered the same issue as you. Could you please share how you resolved it? Any guidance you can provide would be appreciated. Thank you.

jacqueshsu, 2024-01-09:
Which interface for your network?  Wifi or Ethernet?

suktry, 2024-01-09:
> Which interface for your network? Wifi or Ethernet?

wifi

jacqueshsu, 2024-01-09:
I think the issue is related to the WiFi driver. Do you have other wifi card? You can use it to confirm      the root case.

suktry, 2024-01-09:
> I think the issue is related to the WiFi driver. Do you have other wifi card? You can use it to confirm the root case.

I sincerely apologize for not expressing my actual scenario clearly. I am using a virtual machine running Ubuntu, configuring the setup through Bluetooth using chip-tool pairing ble-thread. The hardware device utilized is the esp32h2, and the thread border router is developed with the esp32s3 in conjunction with the esp32h2 development kit. The thread border router connects to the local network via WiFi. In this setup, I can achieve pairing and operations using the Chip application on Android, but the issue persists consistently when using Ubuntu.

goossensbas, 2024-12-21:
Hello,

I have the same issue when trying to commission devices.
I use avahi on my raspberry pi OTBR. 

When i publish a service using openthread command-line on esp32h2, the service is published and found.
Can i conclude that multicast is working and properly routed from Thread network to my test network?

I flashed an esp32C6 with the light example to test the commissioning, but it fails with mdns resolution.
I also tried with a Trdonic LED driver, but it also stops because the mDNS services couldn't be published.

When i run avahi-browse -a it doen't show any matter services.