project-chip/connectedhomeip #27191: [Feature] How to actively unpair/reset connection from Matter device

nmtoan91, 2023-06-12:
### Feature description

...
I want to unpair/reset connection from my Matter device, what should I do?

### Platform

all

### Platform Version(s)

latest

bzbarsky-apple, 2023-06-12:
1. A controller could RemoveFabric the relevant fabrics.
2. The device itself could have UI for removing fabrics; this would be device-specific.
3. The device could have a way to factory reset. This would be device-specific.

closed 2023-06-12 by bzbarsky-apple

nmtoan91, 2023-06-13:
Hi @bzbarsky-apple ,
Can you give a practice sample for removing fabrics/reset? Somehow, my controller fails to remove the fabric and now I cannot pair my device with any controller.

bzbarsky-apple, 2023-06-13:
> Can you give a practice sample for removing fabrics/reset? 

If you are trying to remove your own fabric:

1. Read the CurrentFabricIndex attribute.
2. RemoveFabric with that fabric index.

If you are trying to remove some other fabric, then you would need its fabric index.

You can see an example at https://github.com/project-chip/connectedhomeip/blob/master/src/darwin/CHIPTool/CHIPTool/View%20Controllers/Fabric/FabricUIViewController.m#L453-L489 (though that's an Objective-C example, obviously).

nmtoan91, 2023-07-04:
My scenario is:
My bridged-app was working fine with chip-tool.
Then, I connected the bridged-app to an Apple speaker. Then, I removed it.
Now, I cannot use chip-tool to pair/unpair the bridged-app any more.

Here is the log:

```
[1688432819.213584][1435947:1435947] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1688432819.215405][1435947:1435947] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1688432819.215435][1435947:1435947] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1688432819.215455][1435947:1435947] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1688432819.215603][1435947:1435947] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-CORVJf)
[1688432819.215791][1435947:1435947] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1688432819.215801][1435947:1435947] CHIP:DL: NVS set: chip-counters/reboot-count = 19 (0x13)
[1688432819.216133][1435947:1435947] CHIP:DL: Got Ethernet interface: eno1
[1688432819.216333][1435947:1435947] CHIP:DL: Found the primary Ethernet interface:eno1
[1688432819.216531][1435947:1435947] CHIP:DL: Got WiFi interface: wlp0s20f3
[1688432819.218726][1435947:1435947] CHIP:DL: Found the primary WiFi interface:wlp0s20f3
[1688432819.218775][1435947:1435947] CHIP:IN: UDP::Init bind&listen port=0
[1688432819.218820][1435947:1435947] CHIP:IN: UDP::Init bound to port=34054
[1688432819.218839][1435947:1435947] CHIP:IN: UDP::Init bind&listen port=0
[1688432819.218875][1435947:1435947] CHIP:IN: UDP::Init bound to port=40211
[1688432819.218880][1435947:1435947] CHIP:IN: BLEBase::Init - setting/overriding transport
[1688432819.218884][1435947:1435947] CHIP:IN: TransportMgr initialized
[1688432819.218893][1435947:1435947] CHIP:FP: Initializing FabricTable from persistent storage
[1688432819.218980][1435947:1435947] CHIP:TS: Last Known Good Time: 2023-05-10T09:39:50
[1688432819.219466][1435947:1435947] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xC2E251AC1A70B6C1, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1688432819.222186][1435947:1435947] CHIP:ZCL: Using ZAP configuration...
[1688432819.228241][1435947:1435947] CHIP:CTL: System State Initialized...
[1688432819.228255][1435947:1435947] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1688432819.228264][1435947:1435947] CHIP:CTL: Setting attestation nonce to random value
[1688432819.228271][1435947:1435947] CHIP:CTL: Setting CSR nonce to random value
[1688432819.228282][1435947:1435947] CHIP:IN: UDP::Init bind&listen port=5550
[1688432819.228315][1435947:1435947] CHIP:IN: UDP::Init bound to port=5550
[1688432819.228319][1435947:1435947] CHIP:IN: UDP::Init bind&listen port=5550
[1688432819.228363][1435947:1435947] CHIP:IN: UDP::Init bound to port=5550
[1688432819.228367][1435947:1435947] CHIP:IN: TransportMgr initialized
[1688432819.228509][1435947:1435949] CHIP:DL: CHIP task running
[1688432819.228546][1435947:1435949] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1688432819.228659][1435947:1435949] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1688432819.228667][1435947:1435949] CHIP:CTL: Setting attestation nonce to random value
[1688432819.228704][1435947:1435949] CHIP:CTL: Setting CSR nonce to random value
[1688432819.228995][1435947:1435949] CHIP:CTL: Generating NOC
[1688432819.229249][1435947:1435949] CHIP:FP: Validating NOC chain
[1688432819.229646][1435947:1435949] CHIP:FP: NOC chain validation successful
[1688432819.229683][1435947:1435949] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1688432819.229689][1435947:1435949] CHIP:TS: Last Known Good Time: 2023-05-10T09:39:50
[1688432819.229693][1435947:1435949] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1688432819.229696][1435947:1435949] CHIP:TS: Retaining current Last Known Good Time
[1688432819.230243][1435947:1435949] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1688432819.230584][1435947:1435949] CHIP:TS: Committing Last Known Good Time to storage: 2023-05-10T09:39:50
[1688432819.230885][1435947:1435949] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: C2E251AC1A70B6C1)
[1688432819.230892][1435947:1435949] CHIP:IN: UDP::Init bind&listen port=5550
[1688432819.230928][1435947:1435949] CHIP:IN: UDP::Init bound to port=5550
[1688432819.230933][1435947:1435949] CHIP:IN: UDP::Init bind&listen port=5550
[1688432819.230946][1435947:1435949] CHIP:IN: UDP::Init bound to port=5550
[1688432819.230949][1435947:1435949] CHIP:IN: TransportMgr initialized
[1688432819.233747][1435947:1435949] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000457]
[1688432819.233754][1435947:1435949] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1688432819.233761][1435947:1435949] CHIP:DIS: OperationalSessionSetup[1:0000000000000457]: State change 1 --> 2
[1688432819.434654][1435947:1435949] CHIP:DIS: Checking node lookup status after 201 ms
[1688432827.781085][1435947:1435948] CHIP:DL: Got IP address on interface: wlp0s20f3 IP: 192.168.11.26
[1688432827.781230][1435947:1435949] CHIP:DL: HandlePlatformSpecificBLEEvent 32770
[1688432850.246843][1435947:1435949] CHIP:DIS: Timeout waiting for mDNS resolution.
[1688432851.642906][1435947:1435948] CHIP:DL: Got IP address on interface: wlp0s20f3 IP: 192.168.11.26
[... cut ...]
[1688432864.247318][1435947:1435947] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1688432864.247354][1435947:1435947] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1688432864.247367][1435947:1435947] CHIP:DL: Inet Layer shutdown
[1688432864.247377][1435947:1435947] CHIP:DL: BLE shutdown
[1688432864.247388][1435947:1435947] CHIP:DL: System Layer shutdown
[1688432864.247532][1435947:1435947] CHIP:TOO: Run command failure: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
```

nmtoan91, 2023-07-04:
@bzbarsky-apple Can you take a look and suggest to me what should I do?
Is there any option to reset from the bridged matter device side?

nmtoan91, 2023-07-04:
I tried this one but it not workds
```
chip::Server::GetInstance().ScheduleFactoryReset();
ApplicationInit();
```

bzbarsky-apple, 2023-07-07:
Sorry for the lag; I was on vacation for a bit over a week, but github has no good way to mark that.

>  Then, I connected the bridged-app to an Apple speaker.

How exactly did you do this bit?  Did you open a commissioning window from chip-tool?  Something else?

>  Then, I removed it.

Again: how exactly was this part done?

> Here is the log:

From running which chip-tool command?

nmtoan91, 2023-07-12:
@bzbarsky-apple, Sorry to bother you again. I still cannot pair my device to chip-tool.
This is the full log of my bridged-device:
```
./out/debug/chip-bridge-app
[1689144294.619974][3394940:3394940] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1689144294.620319][3394940:3394940] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1689144294.620330][3394940:3394940] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
onNewNodeProfile
Controller found.
[1689144294.620600][3394940:3394940] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[TOANSTT] Requesting NodeInstanceListS

[1689144294.620729][3394940:3394940] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1689144294.620796][3394940:3394940] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1689144294.620999][3394940:3394940] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-h1rztq)
[1689144294.621181][3394940:3394940] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1689144294.621201][3394940:3394940] CHIP:DL: NVS set: chip-counters/reboot-count = 1 (0x1)
[1689144294.621284][3394940:3394940] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Y4xEgz)
[1689144294.621484][3394940:3394940] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1689144294.621503][3394940:3394940] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1689144294.621560][3394940:3394940] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-5cvU0U)
[1689144294.621742][3394940:3394940] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1689144294.621763][3394940:3394940] CHIP:DL: NVS set: chip-counters/boot-reason = 0 (0x0)
[1689144294.621868][3394940:3394940] CHIP:DL: writing settings to file (/tmp/chip_config.ini-3fL8OX)
[1689144294.622027][3394940:3394940] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1689144294.622044][3394940:3394940] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
[1689144294.622134][3394940:3394940] CHIP:DL: writing settings to file (/tmp/chip_config.ini-wHUNI0)
[1689144294.622313][3394940:3394940] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1689144294.622335][3394940:3394940] CHIP:DL: NVS set: chip-config/location-capability = 2 (0x2)
[1689144294.622789][3394940:3394940] CHIP:DL: Got Ethernet interface: eno1
[1689144294.623137][3394940:3394940] CHIP:DL: Found the primary Ethernet interface:eno1
[1689144294.623468][3394940:3394940] CHIP:DL: Got WiFi interface: wlp0s20f3
[1689144294.625408][3394940:3394940] CHIP:DL: Found the primary WiFi interface:wlp0s20f3
[1689144294.625425][3394940:3394940] CHIP:SPT: *** WARNING: Using temporary passcode 20202021 due to no neither --passcode or --spake2p-verifier-base64 given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure onboarding credentials. ***
[1689144294.625449][3394940:3394940] CHIP:SPT: *** WARNING: Using temporary test discriminator 3840 due to --discriminator not given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure discriminator. ***
[1689144294.625457][3394940:3394940] CHIP:SPT: PASE PBKDF iterations set to 1000
[1689144294.625467][3394940:3394940] CHIP:SPT: LinuxCommissionableDataProvider didn't get a PASE salt, generating one.
[1689144294.628706][3394940:3394940] CHIP:DL: Device Configuration:
[1689144294.628724][3394940:3394940] CHIP:DL:   Serial Number: TEST_SN
[1689144294.628735][3394940:3394940] CHIP:DL:   Vendor Id: 65521 (0xFFF1)
[1689144294.628746][3394940:3394940] CHIP:DL:   Product Id: 32769 (0x8001)
[1689144294.628752][3394940:3394940] CHIP:DL:   Product Name: TEST_PRODUCT
[1689144294.628762][3394940:3394940] CHIP:DL:   Hardware Version: 0
[1689144294.628774][3394940:3394940] CHIP:DL:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
[1689144294.628786][3394940:3394940] CHIP:DL:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
[1689144294.628805][3394940:3394940] CHIP:DL:   Manufacturing Date: (not set)
[1689144294.628817][3394940:3394940] CHIP:DL:   Device Type: 65535 (0xFFFF)
[1689144294.628828][3394940:3394940] CHIP:-: ==== Onboarding payload for Standard Commissioning Flow ====
[1689144294.628849][3394940:3394940] CHIP:SVR: SetupQRCode: [MT:-24J042C00KA0648G00]
[1689144294.628875][3394940:3394940] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1689144294.628886][3394940:3394940] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J042C00KA0648G00
[1689144294.628916][3394940:3394940] CHIP:SVR: Manual pairing code: [34970112332]
[1689144294.628956][3394940:3394940] CHIP:SVR: Initializing subscription resumption storage...
[1689144294.629134][3394940:3394940] CHIP:DL: writing settings to file (/tmp/chip_kvs-VXCsGn)
[1689144294.629404][3394940:3394940] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
[1689144294.629427][3394940:3394940] CHIP:SVR: Server initializing...
[1689144294.629437][3394940:3394940] CHIP:FP: Initializing FabricTable from persistent storage
[1689144294.629483][3394940:3394940] CHIP:TS: Last Known Good Time: 2023-07-12T15:44:12
[1689144294.629531][3394940:3394940] CHIP:DMG: AccessControl: initializing
[1689144294.629539][3394940:3394940] CHIP:DMG: Examples::AccessControlDelegate::Init
[... cut ...]
[1689145247.883668][156580:156580] CHIP:DL: Inet Layer shutdown
[1689145247.883691][156580:156580] CHIP:DL: BLE shutdown
[1689145247.883724][156580:156580] CHIP:BLE: CancelBleIncompleteConnection() failed, err = src/ble/BleLayer.cpp:372: CHIP Error 0x00000003: Incorrect state
[1689145247.883745][156580:156580] CHIP:DL: System Layer shutdown
[1689145247.883978][156580:156580] CHIP:TOO: Run command failure: ../connectedhomeip/examples/chip-tool/commands/common/CHIPCommand.cpp:549: CHIP Error 0x00000032: Timeout
```

bzbarsky-apple, 2023-07-12:
@nmtoan91 I asked three questions; you answered 1...

That said, what those last logs show is that the bridge-app is advertising but chip-tool does not seem to see those advertisements.  You should test whether something like avahi (on Linux) or dns-sd (on Mac) sees the advertisements, for a start...