project-chip/connectedhomeip #27198: How to implement tampering in the matter specification？

seagullanddog, 2023-06-12:
"tamper" in the zigbee specification, the cluster id is 0x0500, but no similar cluster id is found in the matter specification, how does it solve the problem of tamper alarm?

fuxiaoming-lumi, 2023-06-12:
The General Diagnostics cluster has an attribute ActiveHardwareFaults which contains `TemperDetected`.

bzbarsky-apple, 2023-06-12:
Not an SDK issue.

closed 2023-06-12 by bzbarsky-apple

seagullanddog, 2023-06-13:
@fuxiaoming-lumi thank you for your answer. When the "ActiveHardwareFaults" attribute is subscribed, the following code is called, but the attribute value reported to chip-tool is still empty. Why is this, and I hope to get your help again, thanks again?

**_" 
  GeneralFaults<kMaxHardwareFaults> previous;
  GeneralFaults<kMaxHardwareFaults> current;
  ReturnOnFailure(previous.add(to_underlying(EMBER_ZCL_HARDWARE_FAULT_TYPE_RADIO)));
  ReturnOnFailure(previous.add(to_underlying(EMBER_ZCL_HARDWARE_FAULT_TYPE_POWER_SOURCE)));
  ReturnOnFailure(current.add(to_underlying(EMBER_ZCL_HARDWARE_FAULT_TYPE_RADIO  )));
  ReturnOnFailure(current.add(to_underlying(EMBER_ZCL_HARDWARE_FAULT_TYPE_SENSOR)));
  ReturnOnFailure(current.add(to_underlying(EMBER_ZCL_HARDWARE_FAULT_TYPE_POWER_SOURCE)));
  current.add(to_underlying(EMBER_ZCL_HARDWARE_FAULT_TYPE_TAMPER_DETECTED)); "_**

bzbarsky-apple, 2023-06-13:
@seagullanddog Are you also calling `Clusters::GeneralDiagnosticsServer::Instance().OnHardwareFaultsDetect(previous, current);`?

And are you changing the value that `GetActiveHardwareFaults()` returns to reflect the new value?

seagullanddog, 2023-06-14:
@bzbarsky-apple , I don't think calling the GetActiveHardwareFaults() function has any effect, because its content does not return anything, and its code is as follows:

**_CHIP_ERROR DiagnosticDataProviderImpl::GetActiveHardwareFaults(GeneralFaults<kMaxHardwareFaults> & hardwareFaults)
{
#if CHIP_CONFIG_TEST
    ReturnErrorOnFailure(hardwareFaults.add(EMBER_ZCL_HARDWARE_FAULT_TYPE_RADIO));
    ReturnErrorOnFailure(hardwareFaults.add(EMBER_ZCL_HARDWARE_FAULT_TYPE_SENSOR));
    ReturnErrorOnFailure(hardwareFaults.add(EMBER_ZCL_HARDWARE_FAULT_TYPE_POWER_SOURCE));
    ReturnErrorOnFailure(hardwareFaults.add(EMBER_ZCL_HARDWARE_FAULT_TYPE_USER_INTERFACE_FAULT));
#endif

    return CHIP_NO_ERROR;
}_**

bzbarsky-apple, 2023-06-14:
>  I don't think calling the GetActiveHardwareFaults() function has any effec

No, you need to _implement_ GetActiveHardwareFaults to return the actual state of your device (in fact, you need to implement the entire DiagnosticDataProvider).  The code you are citing is from some default test-only implementation, which you should not be using in production.

seagullanddog, 2023-06-16:
@bzbarsky-apple thank you，I have implemented chio-tool to read the ActiveHardwareFaults attribute with a value of 10 (EMBER_ZCL_HARDWARE_FAULT_TYPE_TAMPER_DETECTED). I join the device to the Home pod mini. The Home app of the iPhone does not see the "Tamper" alarm, is it because the iPhone app does not yet support the "Tamper" alarm display?

bzbarsky-apple, 2023-06-16:
>  is it because the iPhone app does not yet support the "Tamper" alarm display?

Probably, yes.