project-chip/connectedhomeip #27210: The NSLocalizedString() in MTRError:errorForCHIPErrorCode may crash when Invoke from Matter extension

huangjinlong92568, 2023-06-13:
When Invoke Matter API from Matter extension, the bundle maybe nil for NSLocalizedString(), in this situation, the errorForCHIPErrorCode API may cause a crash for the `[userInfo addEntriesFromDictionary:@{
            NSLocalizedDescriptionKey : NSLocalizedString(@"The device is already a member of this fabric.", nil)
        }];` (for example)

huangjinlong92568, 2023-06-13:
the bundle maybe nil  because we swizzed the `localizedStringForKey:value:table:` method for NSBundle

ksperling-apple, 2023-06-13:
The return value of `[NSBundle localizedStringForKey:value:table:]` is not nullable, you need to ensure you're returning a string if you're swizzling it.

closed 2023-06-13 by ksperling-apple