project-chip/connectedhomeip #27213: [BUG] TE1 V1.2  TC-S-1.1 and 2.1 : Missing implementation of scene-table-size and remaining-capacity attribute of scenes cluster

VaishaliAvhale, 2023-06-13:
### Reproduction steps

1. Commission a device
2. Run the below command
          ./chip-tool scenes read attribute-list 1 1
As SceneTableSize and RemainingCapacity attribute is mandatory expected its entries present in output - 
- S.S.A0006(SceneTableSize) 
- S.S.A0007(RemainingCapacity)
but it's missing!!!
[esp32c3.txt](https://github.com/project-chip/connectedhomeip/files/11733650/esp32c3.txt)
[chip_tool.txt](https://github.com/project-chip/connectedhomeip/files/11733651/chip_tool.txt)

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

core

### Platform Version(s)

v1.2

bzbarsky-apple, 2023-06-13:
>  but it's missing!!!

@VaishaliAvhale did you enable those attributes in the ZAP configuration?

lpbeliveau-silabs, 2024-08-21:
Closing as those attributes are outdate and now available in the current scenes cluster version.

closed 2024-08-21 by lpbeliveau-silabs