project-chip/connectedhomeip #27225: [BUG]

aravindone, 2023-06-14:
### Reproduction steps

Dear all i facing failure while commissioning the esp 32 with chip-tool through mobile hotspot. i would also need to know when i try using data card the communication will established or i should use a typical router for matter commissioning ?  please comment @dhrishi @bzbarsky-apple   
[mobile hotspot failure.txt](https://github.com/project-chip/connectedhomeip/files/11742646/mobile.hotspot.failure.txt)

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

bfa05804ac3c224c9e70921a857ffcac32ccc7f4

### Platform

esp32

shubhamdp, 2023-06-14:
@aravindone `ConnectNetworkResponse` received a status response 5 i.e `NetworkNotFound Cannot find AP: SSID Not found`.

Are you using 5GHz WiFi network? ESP32 only works with 2.4GHz wifi network. Can you try with any 2.4GHz network?

aravindone, 2023-06-14:
@shubhamdp  the hot spot bandwidth i provided is 2.4GHz and i even given try giving the ssid and password in hex.
the clarification i need is whether this problem will exist in data card i mean some portable modems or we should move with router ?

bzbarsky-apple, 2023-06-14:
Nothing in Matter cares about where your network is coming from, really, as long as the network exists and can handle DNS-SD.

But in this case the esp32 is unable to join the network involved, and you should probably investigate why.

aravindone, 2023-06-26:
The issue was resolving after matter communication established over Datacard

closed 2023-06-26 by aravindone