project-chip/connectedhomeip #27241: [TIMESYNC] SDK has to be implemented for Time Synchronization Cluster

Survensa, 2023-06-14:
### Reproduction steps

**All-clusters-app has to be enabled with below mentioned Attributes :**
- [ ] UTCTime
- [ ] Granularity
- [ ] TimeSource
- [ ] TrustedTimeSource
- [ ] DefaultNTP
- [ ] TimeZone
- [ ] DSTOffset
- [ ] LocalTime
- [ ] TimeZoneDatabase
- [ ] NTPServerAvailable
- [ ] TimeZoneListMaxSize
- [ ] DSTOffsetListMaxSize
- [ ] SupportsDNSResolve

All-clusters-app has to be enabled with below mentioned commands :
- [ ] SetUTCTime
- [ ] SetTrustedTimeSource
- [ ] SetTimeZone
- [ ] SetTimeZoneResponse
- [ ] SetDSTOffset
- [ ] SetDefaultNTP

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

40fb7c246671e9d37ca8bb2cfbf37f81c1509334

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

cjandhyala, 2023-06-28:
PR  #26082 to address this

cecille, 2023-07-17:
The linked PR has now been merged - all clusters should have time sync support now.

closed 2023-07-17 by cecille