project-chip/connectedhomeip #27250: [BUG] No connection with matter device based on ESP32-S3 with W5500 Ethernet IC

mgoeller, 2023-06-14:
### Reproduction steps

I just compiled a matter firmware with esp-idf 5.1 (same with stable 5.0.2) and the latest connectedhomeip example for esp32 for a single lightbul. Using ESP32-S3 as hardware and a W5500 ethernet chip as indicated before.

I can pair with Google Home, however the device goes immediately offline after successful pairing. I can from there still share to Apple Home (as another fabric) and everything works fine there. Immediately pairing with Apple Home works as well, and stays online. Sharing to Google Home works, but goes offline right after.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

connectedhomeip

### Platform

esp32

bzbarsky-apple, 2023-06-14:
@dhrishi @shubhamdp

PSONALl, 2023-06-15:
@mgoeller This is  the issue with Google Home, I have reported it to Google's Issue Tracker
I will notify you whenever the next update is available

mgoeller, 2023-06-16:
Can we somehow emphasis this issue?

ajit803, 2024-05-27:
I can see that the issue is still present in Google Home as of today.

dhrishi, 2024-10-25:
@mgoeller Can you please confirm if this works fine now? If yes, please close the issue