project-chip/connectedhomeip #27258: [BUG] KeyPairDelegate for Android/Java is not correctly documented

maurycyw, 2023-06-14:
### Reproduction steps

We are implementing matter pairing on Android with an external operationalKeyStore which essentially means we need to initialize the chip controller with the NOC, root certificate, and keypairDelegate so that it can pair devices a newly created fabric. 

The public key returned from KeyPairDelegate#getPublicKey: 
```
    /**
     * Returns the DER-encoded X.509 public key, generating a new private key if one has not already
     * been created.
     *
     * @throws KeypairException if a public key could not be resolved
     */
    override fun getPublicKey(): ByteArray {
```
is being compared to the public key in the NOC and failing when the controller is being instantiated.

Note: I verified that the public keys are identical via openssl

After much debugging I found that the NOC public key is being read in an `uncompressed` format but the publicKey we are providing in the delegate is a `compressed` public key. 

The docs were misleading in that the public key that we are supposed to return should be that of a uncompressed EC point. (unless I am not understanding something)

### Bug prevalence

Every time I try to create a chip controller

### GitHub hash of the SDK that was being used

5a21d17fd2bb0a48f4a356937f4741f6c9dc7975

### Platform

android

yunhanw-google, 2023-07-27:
@yufengwangca  could you update the documents for keyPairDelegate? thanks

maurycyw, 2023-08-28:
Note: This could also be related to the `getEncoded()` implementations for the java security apis and which bouncy castle impl is used for ECPublic/Private key. I have since migrated over to spongyCastle but the docs are still misleading so this is still a valid issue since regardless of the implementation the user needs to return an uncompressed point.

aroraprateek202112, 2023-12-21:
Hi @maurycyw ,
I am trying to initialize the Chip controller using the keypair delegate. I am getting a callback on getPublicKey() method of KeypairDelegate. But when I returned myPublicKey.getEncoded() from getPublicKey() method. I am getting the below error and app is getting crashed.

Logs:

> **Failed NOC chain validation: src/credentials/FabricTable.cpp:311: CHIP Error 0x00000050: Unsupported certificate format**
> Reverting pending fabric data for fabric 0x1
> Reverting pending fabric data for fabric 0x1
> Pending Last Known Good Time: 2023-10-14T01:16:48
> KVS: Getting key g/lkgt
> Previous Last Known Good Time: 2023-10-14T01:16:48
> Reverted Last Known Good Time to previous value
> getDeviceController ex :
> **chip.devicecontroller.ChipDeviceControllerException: src/credentials/FabricTable.cpp:311: CHIP Error 0x00000050: Unsupported certificate format
> at chip.devicecontroller.ChipDeviceController.newDeviceController(Native Method)**

I have tried the **toUncompressedPoint()** method but still getting the same error.

I am using the spongycastle to create the Keypair and CSR. Please find the code below:

````
    private val KEY_ALIAS = "Test_Matter_Key"
    private val CN_PATTERN =
        "CN=ABC, O=ABC India, OU=IIC, L=Bengaluru, ST=Karnataka, C=INDIA, emailAddress=aroraprateek202112@gmail.com"

    /**
     * Generate the desired keypair
     *
     * @param alg
     * @param keySize
     * @return
     */
    fun generateKeyPair(keySize: Int): KeyPair? {
        try {
            val ks: KeyStore = KeyStore.getInstance("AndroidKeyStore").apply { load(null) }
            var keyPair: KeyPair? = null
            if (!ks.containsAlias(KEY_ALIAS)) {
                val keyPairGenerator: KeyPairGenerator = KeyPairGenerator.getInstance(
                    KeyProperties.KEY_ALGORITHM_EC,
                    "AndroidKeyStore"
                )
                keyPairGenerator.initialize(
                    KeyGenParameterSpec.Builder(
                        KEY_ALIAS,
                        KeyProperties.PURPOSE_SIGN or KeyProperties.PURPOSE_VERIFY
                    )
                        .setKeySize(keySize)
                        .build()
                )
                keyPair = keyPairGenerator.generateKeyPair()
            } else {
                val entry = ks.getEntry(KEY_ALIAS, null)
                if (entry !is PrivateKeyEntry) {
                    return null
                }
                val privateKey = entry.privateKey
                val publicKey = ks.getCertificate(KEY_ALIAS).publicKey
                keyPair = KeyPair(publicKey, privateKey)
            }
            return keyPair
        } catch (ex: Exception) {
            ex.printStackTrace()
        }
        return null
    }

    /**
     * Generate the desired CSR for signing
     *
     * @param sigAlg
     * @param keyPair
     * @return
     */
    fun generateCSR(keyPair: KeyPair): String {

        val pk10Builder: PKCS10CertificationRequestBuilder = JcaPKCS10CertificationRequestBuilder(
            X500Name(CN_PATTERN),
            keyPair.public
        )
        val csBuilder = JcaContentSignerBuilder("SHA256withECDSA")
        val signer = csBuilder.build(keyPair.private)
        val csr = pk10Builder.build(signer)
        return getCSRStringFromByteArray(csr.encoded)
    }

    private fun getCSRStringFromByteArray(encodedBytes: ByteArray): String {
        val pemObject = PemObject("CERTIFICATE REQUEST", encodedBytes)
        val stringWriter = StringWriter()
        val pemWriter = PemWriter(stringWriter)
        pemWriter.writeObject(pemObject)
        pemWriter.close()
        stringWriter.close()
        return stringWriter.toString()
    }
````

Could you guide me what could be wrong either my KeyPair generation logic or passing of publicKey.getEncoded() in KeyPairDelegate or anything else which is causing issue in intializing the Chip controller.

maurycyw, 2023-12-21:
@aroraprateek202112 I have not encountered the `Unsupported certificate format` error myself but also I am not using the entirety of the KeyPairDelegate.

However given your error, I would export the certificate and use openssl to verify that you are creating a valid certificate (ex: `openssl ec -in my-private-key.pem -noout -text `). 

Only other difference I see is that you are using the Android Keystore. I am not too familiar with it but do you not need to define: `setDigests(KeyProperties.DIGEST_SHA256, KeyProperties.DIGEST_SHA512)` ? Also not sure why you must specify a key size. Essentially I am just not sure if the key returned back from the Keystore is in the expected format so I would test this. 

Apologies I cant be more helpful.

aroraprateek202112, 2023-12-22:
@maurycyw Thanks for you quick reply.

If you have any sample of expected public key format, please share it. I can use that as reference to validate my generated key.

OlivierGre, 2024-02-06:
> external operationalKeyStore which essentially means we need to initialize the chip controller with the NOC, root certificate, and keypairDelegate so that it can pair devices to a newly created fabric.

@yunhanw-google Can you please point me the documentation explaining how to use keypairDelegate and how to set an external operationalKeyStore?

Thanks!

maurycyw, 2024-02-07:
@aroraprateek202112 I don't have an example handy but there is an example of minimum required fields in the [Matter spec](https://csa-iot.org/wp-content/uploads/2022/11/22-27349-001_Matter-1.0-Core-Specification.pdf) [6.4.7. Node Operational Certificate Signing Request (NOCSR)] for a NOCSR (assuming that is the certificate at fault).