project-chip/connectedhomeip #27261: Re-Architect Fan Control Server Code and Add Fan Control Delegate

mhazley, 2023-06-15:
### Description

From chats with various engineers, we discussed how the overall design goal was to:

1. Each cluster instance (i.e. endpoint, cluster pair) is represented by an object.
2. That object has some built-in behavior and some ways for the application to customize its behavior as needed.
3. There is no weird global function that then has to do dispatch going on.

The Fan Control Server currently relies on the global attribute callback function for the implementing device to get attribute changes so we need to improve this by offering some kind of Fan Delegate class description that implementers can use when instantiating. 

The `Step` command was also introduced for 1.2 and its effect is entirely application specific so we need a way to offload this to the application, and the above delegate idea would be perfect for this.

### Task List

- [ ] Re-architect Fan Control server to use `CommandHandlerInterface` and `AttributeAccessInterface`
- [ ] Design a virtual FanDelegate to handle writes and commands
- [ ] Add to all-clusters-app

bzbarsky-apple, 2023-06-15:
>  Re-architect Fan Control server to use CommandHandlerInterface and AttributeAccessInterface

Note that this is not necessarily a prerequisite for doing the FanDelegate thing.  It depends on where you want the responsibility for the attribute data to live, in terms of whether you use AttributeAccessInterface.

As for CommandHandlerInterface, that's not a prerequisite for anything, really.

mhazley, 2023-06-16:
@bzbarsky-apple so are you saying I could keep the server code "as-is" (to a degree...) and still make a delegate that gets called out to by the existing code when writes/commands happen? 

That may require a lot less code churn.

mhazley, 2023-06-16:
it actually seems like following the same pattern as `window-covering-server.cpp` might be a good idea here.

bzbarsky-apple, 2023-06-16:
>  I could keep the server code "as-is" (to a degree...) and still make a delegate that gets called out to by the existing code when writes/commands happe

Yes.

closed 2023-07-18 by mhazley