project-chip/connectedhomeip #27284: [Android arm64] Unable to build chip-tool - unknown argument: '-fmacro-prefix-map=../../='

satyajitanand, 2023-06-16:
### Reproduction steps

Issue Summary: 
Unable to build android arm64 for chip-tool through command line: './scripts/build/build_examples.py --target android-arm64-chip-tool build'

Issue Cause: clang++: error: unknown argument: '-fmacro-prefix-map=../../='

Please find below the error snippet.
2023-06-16 11:36:37 INFO    FAILED: obj/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o 
2023-06-16 11:36:37 INFO    /home/test/Android/Sdk/ndk/21.4.7075529//toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android24-clang++ -MMD -MF obj/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o.d -Wconversion -march=armv8-a -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/mbedtls/repo/include -c ../../src/access/examples/ExampleAccessControlDelegate.cpp -o obj/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o
2023-06-16 11:36:37 INFO    clang++: error: unknown argument: '-fmacro-prefix-map=../../='

I am Sorry, I am unable to upload the build log file (it's not allowing me to upload when tried with error meassge: something went wrong, so I shared the error snippet above).

### Platform

android

### Type

Platform validated

andy31415, 2023-06-16:
Your NDK is 21.4.7075529

Current android dockerfile grabs https://dl.google.com/android/repository/android-ndk-r23c-linux.zip  which seems to be `23.2.8568313`

bzbarsky-apple, 2023-06-16:
Duplicate of #25987.

closed 2023-06-16 by bzbarsky-apple

satyajitanand, 2023-06-19:
Thanks for helping. I think I referred "https://gautesl.github.io/connectedhomeip/docs/guides/android_building.html", which seems to be not updated.
I should have referred "https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md" for Android Matter.