project-chip/connectedhomeip #27287: [TC-DA-1.2] - python script is failing at Test Step 2

Rajashreekalmane, 2023-06-16:
**Summary:**
 TC_DA_1.2 test case is failing while sending AttestationRequest at test step 2 
[image]

**Reproduction steps:**
Setting up the environment : 
Step 1 : cloned Connectedhomeip using  mentioned command : [git clone ](https://github.com/project-chip/connectedhomeip.git) https://github.com/project-chip/connectedhomeip.git
step 2 :  cd connectedhomeip
step 3 : pulled latest commit 
step 4:  source scripts/bootstrap.sh
step 5 :  ./scripts/build_python.sh --install_wheel build-env
step 6 :  source scripts/activate.sh

executed the below mentioned command to put DUT into a commissionable state,
` ./chip-all-clusters-app
`
Once DUT reach the  commissionable state sent below mentioned command on TH.
` rm -f admin_storage.json && python3 TC_DA_1_2.py --discriminator 3840 --passcode 20202021 --commissioning-method on-network --paa-trust-store-path /paa_roots --storage-path admin_storage.json`

**system configuration**
Chip-tool - Linux
DUT - RPI
Commit-ID : 40fb7c246671e9d37ca8bb2cfbf37f81c1509334	
Dut application: all-clusters-app

**PFA logs below**
[TC_DA_1_2.txt](https://github.com/project-chip/connectedhomeip/files/11768609/TC_DA_1_2.txt)
[TC_DA_1.2_DUT.txt](https://github.com/project-chip/connectedhomeip/files/11768610/TC_DA_1.2_DUT.txt)

tcarmelveilleux, 2023-06-16:
This was always working before. There must be something new that got broken from someone else's PR

cecille, 2023-06-16:
The many print errors are coming from a bad merge or something on the update to the matter testing support - print step can now take either a string or an int, but it got changed back by accident.

That all is a red herring though - I believe the real error is here:

  File "/home/grl/May30/connectedhomeip/src/python_testing/TC_DA_1_2.py", line 256, in test_TC_DA_1_2
    asserts.assert_in(vendor_id, range(1, 0xfff0), "Vendor ID is out of range")
  File "/home/grl/May30/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/asserts.py", line 149, in assert_in
    _call_unittest_assertion(_pyunit_proxy.assertIn,
  File "/home/grl/May30/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/asserts.py", line 52, in _call_unittest_assertion
    raise signals.TestFailure(my_msg, extras=extras)

cecille, 2023-06-16:
AKA, I think you're using a test vendor ID. You can bypass that check for dev and in the CI by using the PICS_SDK_CI_ONLY PICS, but for real cert the vendor ID needs to be the manufacturer assigned vendor ID, which is not in that range

Rajashreekalmane, 2023-06-20:
Above described issue resolved in latest Commit : f5c462133547c290f7eb09e5628ede162fb4a218

But it still failing at Test Step 8.3 : Verify mandatory cd contents

```
[MatterTest] 06-20 11:53:14.182 INFO ***** Test Step 8 : Verify that the attestation_elements_message structure fields satisfy the following conditions
[MatterTest] 06-20 11:53:14.182 INFO ***** Test Step 8.1 : Verify attestation elements size is < = 900 bytes
[MatterTest] 06-20 11:53:14.182 INFO ***** Test Step 8.2 : Verify certification declaration is present and follows spec format
[MatterTest] 06-20 11:53:14.183 INFO ***** Test Step 8.3 : Verify mandatory cd contents
[MatterTest] 06-20 11:53:14.189 ERROR Exception occurred in test_TC_DA_1_2.
Traceback (most recent call last):
  File "/home/grl/Jun14/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/base_test.py", line 777, in exec_one_test
    test_method()
  File "/home/grl/Jun14/connectedhomeip/src/python_testing/matter_testing_support.py", line 933, in async_runner
    return asyncio.run(body(*args, **kwargs))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/grl/Jun14/connectedhomeip/src/python_testing/TC_DA_1_2.py", line 256, in test_TC_DA_1_2
    asserts.assert_in(vendor_id, range(1, 0xfff0), "Vendor ID is out of range")
  File "/home/grl/Jun14/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/asserts.py", line 149, in assert_in
    _call_unittest_assertion(_pyunit_proxy.assertIn,
  File "/home/grl/Jun14/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/asserts.py", line 52, in _call_unittest_assertion
    raise signals.TestFailure(my_msg, extras=extras)
mobly.signals.TestFailure: Details=65521 not found in range(1, 65520) Vendor ID is out of range, Extras=None
[MatterTest] 06-20 11:53:14.191 INFO [Test] test_TC_DA_1_2 FAIL
[MatterTest] 06-20 11:53:14.194 INFO Summary for test class TC_DA_1_2: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0
[MatterTest] 06-20 11:53:14.195 INFO Summary for test run MatterTest@06-20-2023_11-53-12-974:
```
**PFA logs below**
[TC_DA_1_2.txt](https://github.com/project-chip/connectedhomeip/files/11796659/TC_DA_1_2.txt)

closed 2023-06-23 by Rajashreekalmane