project-chip/connectedhomeip #27288: Matter devices not working if Mobile is in different network

sahityadubey, 2023-06-16:
### Reproduction steps

I have commissioned my Google nest and matter device.
When I am changing my mobile network, I am not able to communicate with my POC app. but Google Home is able to communicate with the Matter device. 

Can I understand how my matter device can work in different network.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

TE_23_02/rc2

### Platform

android

andy31415, 2023-06-16:
Also the bug details are quite vague (not sure if this is on purpose or not) ... I do not know what the proof of concept application is or what APIs it is using.

tcarmelveilleux, 2023-06-16:
@sahityadubey Can you give more details about which app you are using? Which exact repos path is it that you are building?

sahityadubey, 2023-06-16:
I would like to present a specific use case that I am currently facing with regard to controlling Matter accessories, specifically a Smart Bulb, using a Google Nest Hub and an Android app named TestApp developed with the Chiptool SDK.

The process I followed is as follows:

Commissioned a Google Nest Hub in the Google Home app.
Commissioned a Matter accessory (Smart Bulb) into the Google Home app.
Developed an Android app named TestApp using the Chiptool SDK.
Shared the Matter device with my TestApp using a QR code.
Intended to control (Turn On/Off) the Matter accessory from the TestApp.
However, I have encountered an issue in the following scenario: When attempting to control the Matter accessory from the TestApp on a different network, specifically when the mobile device is connected to a network separate from the one where the Google Nest Hub and Matter accessory are connected, an error is thrown.

Given this situation, I am seeking guidance on whether there is a method to control Matter accessories from a different network. I would greatly appreciate your assistance in resolving this matter and enabling seamless control of the Matter accessory across different networks.