project-chip/connectedhomeip #27305: [ESP32] Fails the matter test case (TC-OO-2.3)

maenkai, 2023-06-17:
### Reproduction steps

esp-idf: v4.4.3
esp-matter: 2f0929e5f8fd58b1a4cbaa2bd8b171ce50ad2a75 (HEAD -> main, origin/main, origin/HEAD)

Test TC-OO-2.3 using TH2.6 tool 
- offWaitTime: 

- Test Failure: value value < minValue: 167 < 170

```c
TC-O-2.3
12b: Reads OffWaitTime attribute from DUT
21b: Reads OffWaitTime attribute from DUT
24c: Reads OffWaitTime attribute from DUT

### Platform

esp32

### Platform Version(s)

2f0929e5f8fd58b1a4cbaa2bd8b171ce50ad2a75

### Type

Unit tested

### Anything else?

[_2023_6_17_11_31_28.log](https://github.com/project-chip/connectedhomeip/files/11778006/_2023_6_17_11_31_28.log)

maenkai, 2023-06-17:
[_2023_6_17_11_31_28.log](https://github.com/project-chip/connectedhomeip/files/11778009/_2023_6_17_11_31_28.log)

maenkai, 2023-06-17:
Fail: 12b , 21b , 24c
[image]

VaishaliAvhale, 2023-07-10:
This issue is addressed here 
https://github.com/espressif/esp-matter/issues/336
Please check and close the issue if it is resolved.

dhrishi, 2023-07-11:
Assuming the reporter is the same, I will close this issue. @maenkai Please re-open in case of anything