project-chip/connectedhomeip #27307: [APP-CHIPTOOL]  How does the chiptool app create a device?

ngovancuong2735, 2023-06-19:
### Reproduction steps

I have tried pairing a matter device on 2 different platforms: **1. Google nest hub** and **2. Chip-tool installed on my laptop**. 
I found that on Chip-tool the amount of information exchanged is much less than on Google nest hub. I don't see any important information to create a device on an app like Google such as Device Type, number of endpoints,... So what information is needed to create a device on the app? I have seen a way to get information like BasicInformation or Descriptor. But I don't see other information like the number of endpoints,...

### Bug prevalence

every

### GitHub hash of the SDK that was being used

master

### Platform

other

bzbarsky-apple, 2023-06-20:
@ngovancuong2735 chip-tool does not have a concept of "create a device".  It's a test tool, not a user-oriented controller.

You can get the list of endpoints by reading the PartsList attribute of the Descriptor cluster on endpoint 0.

bzbarsky-apple, 2023-06-20:
Also, I expect Google Nest hub just does a wildcard subscribe, which effectively reads all attributes on the device.