project-chip/connectedhomeip #27313: [BUG]

aravindone, 2023-06-19:
### Reproduction steps

after setting up the environment variable for ESP-IDF.

but setting the Matter target to build im facing stuck 

[set-target_error.txt](https://github.com/project-chip/connectedhomeip/files/11786396/set-target_error.txt)

@shubhamdp @dhrishi please comment

### Bug prevalence

recently now im facing

### GitHub hash of the SDK that was being used

bfa05804ac3c224c9e70921a857ffcac32ccc7f4

### Platform

esp32

shubhamdp, 2023-06-19:
@aravindone Thanks for reporting, will take a look. But, if you are super stuck and want a workaround then please try the following.

```
cd esp-idf && source export.sh
cd connectedhomeip && source scripts/activate.sh
which -a python3 // from the result pick one from IDF environment would be similar to `~/.espressif/python_env/idf4.4_py3.11_env/bin/python3`
~/.espressif/python_env/idf4.4_py3.11_env/bin/python3 `which idf.py` [set-target | build | flash | monitor]
```

aravindone, 2023-07-28:
Hi @shubhamdp  please have a look on this log 
https://raw.githubusercontent.com/aravindone/aravindone/main/esp%20_idf%20error

where im facing some package mismatching

shubhamdp, 2023-08-02:
@aravindone Can you please check the latest steps in here https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/setup_idf_chip.md we have recently updated the IDF to v5.1 and gdbgui is not a requirement anymore.

idf-component-manager and click should have been installed, can you try running `source scripts/bootstrap.sh` again?

aravindone, 2023-08-04:
@shubhamdp  after updating the esp-idf to v5.1 and i ran the source scripts/bootstrap.sh  but now when i check the idf.py im getting this error

~/connectedhomeip$ idf.py --version
Please use idf.py only in an ESP-IDF shell environment.

shubhamdp, 2023-08-04:
This is because python package [esp_idf_monitor](https://github.com/project-chip/connectedhomeip/blob/master/scripts/setup/requirements.esp32.txt#L14) is missing.

For, esp-idf v5.1, we have to add few python dependencies, If you haven't bootstrapped after moving the latest connectedhomeip master, then I'd suggest you either install them by bootstrapping again or installing the dependencies from this file.

```
source scripts/bootstrap.sh
# OR

source scripts/activate.sh
python3 -m pip install -r scripts/setup/requirements.esp32.txt -c scripts/setup/constraints.txt
```

aravindone, 2023-08-07:
@shubhamdp 

esp_build error

https://raw.githubusercontent.com/aravindone/aravindone/main/esp_build%20error
im facing this error after running the commands ,please have a look at the log

shubhamdp, 2023-08-07:
Hi @aravindone running `python3 -m pip install -r scripts/setup/requirements.esp32.txt -c scripts/setup/constraints.txt` did not solve the issue?

aravindone, 2023-08-07:
@shubhamdp  yes the issue doesn't solve it shows that kconfgen module is missing ,but when i check it was there

rntbci@inmh100170:~/connectedhomeip/examples/lighting-app/esp32$ pip show kconfgen
Name: kconfgen
Version: 1.2.2
Summary: A generator of (minimal) Linux kernel configuration files.
Home-page: https://github.com/rbarrois/kconfgen
Author: Raphaël Barrois
Author-email: raphael.barrois+kconfgen@polytechnique.org
License: MIT
Location: /home/rntbci/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages
Requires: kconfiglib, toml
Required-by:

shubhamdp, 2023-08-07:
Can you please remove the `kconfiglib` and install `esp-idf-kconfig==1.2.0`.

```
python3 -m pip uninstall kconfiglib
python3 -m pin install esp-idf-kconfig==1.2.0
```
If this does not work, I'd suggest
```
rm -rf .environment`
source scripts/bootstrap.sh
```

aravindone, 2023-08-08:
@shubhamdp  hi tried the above one and set-target was working fine , now the issue is when i build project it shows some error 

https://github.com/aravindone/aravindone/blob/main/esp_build_error

shubhamdp, 2023-08-08:
@aravindone This is version conflict with `pyparsing`. If you you have not done the bootstrap by removing .environment then, please uninstall the `python3 -m pip uninstall pyparsing` and install the version from https://github.com/project-chip/connectedhomeip/blob/master/scripts/setup/constraints.txt#L251 `python3 -m pip install pyparsing==3.0.9`

aravindone, 2023-08-08:
Hi @shubhamdp Thanks for supporting the issue i faced , im  closing the ticket now.

closed 2023-08-08 by aravindone