project-chip/connectedhomeip #27338: [BUG] Bootstrap script fails to run in zsh

umbraHare, 2023-06-20:
### Reproduction steps

1. Open zsh shell.
2. After executing `. ./scripts/bootstrap.sh`, return `_install_additional_pip_requirements:read:19: bad option: -a`.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

9a5b9097245526267909ffbfc7e2f60a3c9b452a

### Platform

other

andy31415, 2023-06-20:
I never tried to use zsh ... I had assumed we just use bash everywhere as our only supported shell.

bzbarsky-apple, 2023-06-20:
The silly part is that what needs to be sourced is the activate part, not the bootstrap part. So maybe we could move the bootstrap bits out into a script that does actually run via bash?

andy31415, 2023-06-20:
ZSH should also have a way to say "split out comma separated string". Looking to see what we can do.

bzbarsky-apple, 2023-06-20:
See the stackoverflow above.  You can `read -r -A` in zsh...