project-chip/connectedhomeip #27361: [BUG] Linux java build fails randomly on Address-ParseOnly Test

woody-apple, 2023-06-21:
### Reproduction steps

https://github.com/project-chip/connectedhomeip/actions/runs/5326895151/jobs/9649546400?pr=27360

Randomly the java linux build seems to timing out on "Running Pairing Address-ParseOnly Test"

Example log is:

```
Tue, 20 Jun 2023 20:47:20 GMT
[2023-06-20 20:47:20.183121][APP ][STDOUT]CHIP:IN: (U) Sending msg 198278943 to IP address 'UDP:[::1%lo]:56518'
Tue, 20 Jun 2023 20:47:20 GMT
[2023-06-20 20:47:20.183222][APP ][STDOUT]CHIP:SC: Sent PBKDF param response
Tue, 20 Jun 2023 21:12:18 GMT
Error: The action has timed out.
```

Odd that it seems potentially locked up.

### Bug prevalence

Pretty often, noticed about 15 failures today

### GitHub hash of the SDK that was being used

top

### Platform

other

woody-apple, 2023-06-21:
Disabled this in this PR: https://github.com/project-chip/connectedhomeip/pull/27395

andy31415, 2025-04-30:
Unassigning myself: I have not had time to work on this for almost 2 years ... unlikely that things will change and keep having this assigned is misleading.