project-chip/connectedhomeip #27379: [BUG] [OTA] event not reported in time

step0035, 2023-06-21:
### Reproduction steps

In TC-SU-2.7
`otasoftwareupdaterequestor subscribe-event state-transition 5  10 0x1234567890 0`

- This step sets the min reporting interval to 5seconds and max reporting interval to 10seconds.
- Once the event is subscribe, execute OTA.
- For some platforms, OTA to prepare for OTA, it needs to erase the flash, which is a blocking call and might exceed the max reporting interval of 10seconds.
- Once the event is missed, the chiptool will not continue to get events from the DUT, causing the test case to fail.
- May I know if this is an expected behaviour?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

83f7a2fe136e0b746db09f1d19e36c693a634b66

### Platform

ameba

bzbarsky-apple, 2023-06-21:
@step0035 Which state is being transitioned into here, exactly?  "Applying"?  Something else?

step0035, 2023-06-22:
@bzbarsky-apple , all of the states are not reporting. Chiptool doesn't even get the periodic subscription report.

bzbarsky-apple, 2023-06-22:
I understand that.  My question was which state transition event, if any would follow immediately after the flash erasure.

That is, if erasing flash took 0 time, what event would you expect to get?

step0035, 2023-06-22:
The first event transition should be Idle (0x01) => Querying (0x02).

bzbarsky-apple, 2023-06-22:
OK.  So how should this work for a real client, not chip-tool, that uses a smallish max-interval?

step0035, 2023-06-22:
Since the subscription is out of sync, to continue to get the subscription reports from server, the client should initiate a re-subscription request.

In my opinion, this is a correct behavior by client/chip-tool. So for TC-SU-2.7 specifically, do you think it could potentially be a test plan issue? Maybe the max-interval should be longer.

bzbarsky-apple, 2023-06-22:
Yes, that would make sense for the test plan.  Which means probably an issue is needed for https://github.com/CHIP-Specifications/chip-test-plans/

step0035, 2023-06-22:
Yes, I think we need to find a way to handle it better rather than changing the test plan. Thanks, I will look into it.

bzbarsky-apple, 2023-06-22:
Did some looking around, and given XIP flash options here might be pretty limited, and in fact the test plan expectation may not be reasonable.

step0035, 2023-06-23:
There's a way to bypass this on platform level. Instead of erasing the whole flash at one go, we can erase one sector at a time before writing a sector to flash.

PR #27513 (2023-07-18): [Ameba] ota fix blocking erase flash and cleanup
- Fix #27379 
- Flash erase will be handled by lower layer and will not block for 20s
- Cleanup code
closed 2023-07-18 by mergify[bot]