project-chip/connectedhomeip #27405: [BUG] Cirque build failed

woody-apple, 2023-06-22:
### Reproduction steps

Noticed this while going through build failures: https://github.com/project-chip/connectedhomeip/actions/runs/5339819857/jobs/9678937869

### Bug prevalence

Saw this once last night

### GitHub hash of the SDK that was being used

509a55cb5d711fa0df339fe135f426190ea0b00d

### Platform

other

woody-apple, 2023-06-26:
Who is the best to look at cirque issues?

andy31415, 2023-06-28:
ToT seems to be mostly green, I see https://github.com/project-chip/connectedhomeip/actions/runs/5401208349/jobs/9810691267 for example and no cirque failures for a while now.

I think we should prioritize depending on occurence count - going through cirque logs is reasonably time-consuming.

woody-apple, 2023-07-07:
Haven't seen one since, closing.

closed 2023-07-07 by woody-apple