project-chip/connectedhomeip #27411: [Build] OpenIOT builds failing occasionally

woody-apple, 2023-06-22:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5343701328/jobs/9687147953

I noticed a couple builds failing like this

### Platform

other

andy31415, 2023-06-22:
Log extract:

```
=================================== FAILURES ===================================
______________________________ test_command_check ______________________________

device = <common.fvp_device.FvpDevice object at 0x7fce765ba880>
...
            if "discriminator" in param_name:
>               assert int(ret[-2].split()[0], 16) == value
E               assert 3840 == 0
E                 +3840
E                 -0

src/test_driver/openiotsdk/integration-tests/shell/test_app.py:166: AssertionError
```

woody-apple, 2023-06-22:
Pinging @ATmobica

ATmobica, 2023-06-22:
@woody-apple It looks like we missed a single byte handling FVP output. I can't find more failure CI with this issue. I want to verify if it always happens in the same place or not. Could you send me more failing builds? You mentioned about few.

woody-apple, 2023-06-23:
@ATmobica here's another: https://github.com/project-chip/connectedhomeip/actions/runs/5336827762/jobs/9672069578

You can see the full list of them here: https://github.com/project-chip/connectedhomeip/actions/workflows/examples-openiotsdk.yaml?query=is%3Afailure

(Note, not all are flakey, some are actually real failures, but you'll need to troll through).

Can I request that if you see an existing issue, feel free to fix that one separately, then look through the rest? Note: I'm keeping a close eye on CI now, and will file any ones that pop up.

ATmobica, 2023-06-26:
@woody-apple Thanks. There are a lot of PRs failures due to provided changes so it's hard to catch the real issues. Anyway, we will monitor all issues and try to fix them continuously. We are still in the upstreaming phase. Our PRs provide some improvements that should guarantee stability and good performance in CI too.
I will be very grateful if you post here all the issues you come across. This will be super helpful.

ATmobica, 2023-06-26:
For example, this PR should provide the CI improvements: https://github.com/project-chip/connectedhomeip/pull/27387

woody-apple, 2023-06-26:
Thanks, closing this. I'll keep monitoring.

closed 2023-06-26 by woody-apple