project-chip/connectedhomeip #27414: [Type Error] TC-FAN-2.5 AirflowDirection attribute type error.

manjunath-grl, 2023-06-22:
### Reproduction steps

Summary:
According to the test-plan and specification, the type of the `AirflowDirection` attribute is `AirflowDirectionEnum`,  but got enum8 with failure response: value type (enum8) is different than the expected type (AirflowDirectionEnum) when executed.

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Test_Fan_2_5_type_Error

Reproduction steps:
Step 1: Add TC-FAN-2.5 test in ciJson file.
Step 2:  Regen
            `./scripts/tools/zap_regen_all.py --type tests`
Step 3:  Build
            ```scripts/run_in_build_env.sh './scripts/build_python.sh --install_wheel build-env --include_yamltests'
                  ./scripts/run_in_build_env.sh \
                     "./scripts/build/build_examples.py \
                        --target linux-x64-chip-tool-no-ble-tsan-clang \
                        --target linux-x64-all-clusters-no-ble-tsan-clang \
                        --target linux-x64-lock-no-ble-tsan-clang \
                        --target linux-x64-ota-provider-no-ble-tsan-clang \
                        --target linux-x64-ota-requestor-no-ble-tsan-clang \
                        --target linux-x64-tv-app-no-ble-tsan-clang \
                        --target linux-x64-bridge-no-ble-tsan-clang \
                        build \"```
Step 4: Run YAML tests
           ```./scripts/run_in_build_env.sh \
                  "./scripts/tests/run_test_suite.py \
                     --chip-tool -with-python \
                     --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool \
                     run \
                     --iterations 1 \
                     --all-clusters-app ./out/linux-x64-all-clusters-no-ble-tsan-clang/chip-all-clusters-app \
                     --lock-app ./out/linux-x64-lock-no-ble-tsan-clang/chip-lock-app \
                     --ota-provider-app ./out/linux-x64-ota-provider-no-ble-tsan-clang/chip-ota-provider-app \
                     --ota-requestor-app ./out/linux-x64-ota-requestor-no-ble-tsan-clang/chip-ota-requestor-app \
                     --tv-app ./out/linux-x64-tv-app-no-ble-tsan-clang/chip-tv-app \
                     --bridge-app ./out/linux-x64-bridge-no-ble-tsan-clang/chip-bridge-app \"```

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

2d78169b34a09e6a613d03f64dbe7c342af44672

### Platform

other

### Anything else?

Spec:
[image]
Test-plan:
[image]
Execution losg:
[image]
[FAN25_Fail_logs.zip](https://github.com/project-chip/connectedhomeip/files/11834882/FAN25_Fail_logs.zip)

bzbarsky-apple, 2023-06-22:
Rhe XML does have AirflowDirectionEnum.

I don't think YAML supports named enum types in its typing, so you have to use the underlying type... @vivien-apple is that correct?

bzbarsky-apple, 2023-06-23:
I just tried reproducing this, and it does not reproduce for me.  The test passes.

manjunath-grl, 2023-06-28:
@vivien-apple Tried on the latest [commit](23a0064eee834ee1e9b0ffc6d243a6a3866c8349) but seeing same failure response: value type (enum8) is different than the expected type (AirflowDirectionEnum) when executed with command `./scripts/tools/zap_regen_all.py`
[image]