project-chip/connectedhomeip #27421: [Build] Darwin test failed

woody-apple, 2023-06-22:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5349369366/jobs/9700494331

```
ERROR   20:39:42.950 - APP  OUT  : [1687466382949] [43734:331988] [EM] Rxd Ack; Removing MessageCounter:95909991 from Retrans Table on exchange 4673r
ERROR   20:39:42.950 - TEST OUT  : 		    Expected: [
ERROR   20:39:42.950 - TEST OUT  : 		                {
ERROR   20:39:42.950 - TEST OUT  : 		                  "values": [
ERROR   20:39:42.950 - TEST OUT  : 		                    {
ERROR   20:39:42.950 - TEST OUT  : 		                      "value": 1
ERROR   20:39:42.950 - TEST OUT  : 		                    }
ERROR   20:39:42.950 - TEST OUT  : 		                  ]
ERROR   20:39:42.950 - TEST OUT  : 		                }
ERROR   20:39:42.950 - TEST OUT  : 		              ]
ERROR   20:39:42.950 - TEST OUT  : 		    Received: [
ERROR   20:39:42.950 - TEST OUT  : 		                 {
ERROR   20:39:42.950 - TEST OUT  : 		                   "attribute": "LockState",
ERROR   20:39:42.950 - TEST OUT  : 		                   "cluster": "DoorLock",
ERROR   20:39:42.950 - TEST OUT  : 		                   "dataVersion": 3671656477,
ERROR   20:39:42.950 - TEST OUT  : 		                   "endpoint": 1,
ERROR   20:39:42.950 - TEST OUT  : 		                   "value": 2
ERROR   20:39:42.950 - TEST OUT  : 		                 }
ERROR   20:39:42.950 - TEST OUT  : 		               ]
ERROR   20:39:42.950 - TEST OUT  : 		***** Test Failure : The test expectation "value == 1" is false
ERROR   20:39:42.950 - TEST OUT  : ✗ Test finished in 56580ms with 27 success, 1 errors and 0 warnings
ERROR   ================ CAPTURED LOG END ====================
ERROR   Test_TC_DRLK_2_4               - FAILED in 70.72 seconds
Traceback (most recent call last):
  File "/Users/runner/work/connectedhomeip/connectedhomeip/./scripts/tests/run_test_suite.py", line 344, in cmd_run
    test.Run(
  File "/Users/runner/work/connectedhomeip/connectedhomeip/scripts/tests/chiptest/test_definition.py", line 340, in Run
    runner.RunSubprocess(
  File "/Users/runner/work/connectedhomeip/connectedhomeip/scripts/tests/chiptest/runner.py", line 167, in RunSubprocess
    raise Exception('Command %r failed: %d' % (cmd, s.returncode))
Exception: Command ['stdbuf', '-o0', '-i0', 'python3', '/Users/runner/work/connectedhomeip/connectedhomeip/scripts/tests/yaml/chiptool.py', 'tests', 'Test_TC_DRLK_2_4', '--PICS', 'src/app/tests/suites/certification/ci-pics-values', '--server_path', './out/darwin-x64-chip-tool-no-ble-tsan-clang/chip-tool', '--server_arguments', 'interactive server --storage-directory /var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/tmpb9l01tio'] failed: 1
Error: Process completed with exit code 2.
```

### Platform

darwin

bzbarsky-apple, 2023-06-23:
And to be clear, the test in this form is not just a problem for CI: it will be a problem for real devices too, leading to random failures on perfectly compliant devices.

manjunath-grl, 2023-06-23:
@bzbarsky-apple @cjandhyala I will update the script back to the old changes in [PR](https://github.com/project-chip/connectedhomeip/pull/27011).

manjunath-grl, 2023-06-23:
@bzbarsky-apple Update please refer below link:
https://github.com/project-chip/connectedhomeip/blob/d3b70e43678b15eba4ea3c0b0c0f67a8ea799add/src/app/tests/suites/certification/Test_TC_DRLK_2_4.yaml

closed 2023-06-23 by mergify[bot]

bzbarsky-apple, 2023-06-23:
@manjunath-grl Yes, thank you, that looks much better.