project-chip/connectedhomeip #27429: [TC-DA-1.2] - python script is failing at Test Step 8.3

Rajashreekalmane, 2023-06-23:
### Reproduction steps

**Reproduction steps:**
Setting up the environment :
Step 1 : cloned Connectedhomeip using mentioned command : [git clone ](https://github.com/project-chip/connectedhomeip.git)https://github.com/project-chip/connectedhomeip.git
step 2 : cd connectedhomeip
step 3 : pulled latest commit
step 4: source scripts/bootstrap.sh
step 5 : ./scripts/build_python.sh --install_wheel build-env
step 6 : source scripts/activate.sh

executed the below mentioned command to put DUT into a commissionable state,
./chip-all-clusters-app
Once DUT reach the commissionable state sent below mentioned command on TH.
 rm -f admin_storage.json && python3 TC_DA_1_2.py --discriminator 3840 --passcode 20202021 --commissioning-method on-network --paa-trust-store-path /paa_roots --storage-path admin_storage.json

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

f5c462133547c290f7eb09e5628ede162fb4a218

### Platform

raspi

### Platform Version(s)

RPI4 8GB RAM

### Anything else?

Test case is failing due to Vendor ID is out of range at test step 8.3 While verifying mandatory cd contents.

```
[MatterTest] 06-20 11:53:14.182 INFO ***** Test Step 8 : Verify that the attestation_elements_message structure fields satisfy the following conditions
[MatterTest] 06-20 11:53:14.182 INFO ***** Test Step 8.1 : Verify attestation elements size is < = 900 bytes
[MatterTest] 06-20 11:53:14.182 INFO ***** Test Step 8.2 : Verify certification declaration is present and follows spec format
[MatterTest] 06-20 11:53:14.183 INFO ***** Test Step 8.3 : Verify mandatory cd contents
[MatterTest] 06-20 11:53:14.189 ERROR Exception occurred in test_TC_DA_1_2.
Traceback (most recent call last):
  File "/home/grl/Jun14/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/base_test.py", line 777, in exec_one_test
    test_method()
  File "/home/grl/Jun14/connectedhomeip/src/python_testing/matter_testing_support.py", line 933, in async_runner
    return asyncio.run(body(*args, **kwargs))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/grl/Jun14/connectedhomeip/src/python_testing/TC_DA_1_2.py", line 256, in test_TC_DA_1_2
    asserts.assert_in(vendor_id, range(1, 0xfff0), "Vendor ID is out of range")
  File "/home/grl/Jun14/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/asserts.py", line 149, in assert_in
    _call_unittest_assertion(_pyunit_proxy.assertIn,
  File "/home/grl/Jun14/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/mobly/asserts.py", line 52, in _call_unittest_assertion
    raise signals.TestFailure(my_msg, extras=extras)
mobly.signals.TestFailure: Details=65521 not found in range(1, 65520) Vendor ID is out of range, Extras=None
[MatterTest] 06-20 11:53:14.191 INFO [Test] test_TC_DA_1_2 FAIL
[MatterTest] 06-20 11:53:14.194 INFO Summary for test class TC_DA_1_2: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0
[MatterTest] 06-20 11:53:14.195 INFO Summary for test run MatterTest@06-20-2023_11-53-12-974:
```

**PFA logs below** : 
[TC_DA_1_2.txt](https://github.com/project-chip/connectedhomeip/files/11796659/TC_DA_1_2.txt)

bzbarsky-apple, 2023-06-23:
> mobly.signals.TestFailure: Details=65521 not found in range(1, 65520) Vendor ID is out of range, Extras=None

Looks like the test is testing that device uses a valid non-test vendor id.  It is guaranteed to fail if you run it against one of the test apps while it's in fact using a test vendor id.

bzbarsky-apple, 2023-06-23:
In particular, the test plan says:

> The vendor_id field is in range [0x0001 .. 0xFFF0]

tcarmelveilleux, 2023-07-18:
Testing in CI uses:

```
is_ci = self.check_pics('PICS_SDK_CI_ONLY')
....

if not is_ci:
            asserts.assert_in(vendor_id, range(1, 0xfff0), "Vendor ID is out of range")
```

Overall, testing against all-clusters-app requires some affordances since it's not a legitimate-looking product.

cecille, 2023-07-18:
+1 to Tennessee - if you're running against a test against the example apps, use a PICS file that includes the PICS_SDK_CI_ONLY PICS. Specify the PICS file as a command line argument to the script as --PICS filename

Rajashreekalmane, 2023-08-31:
As per discussion with @Saravana-kr22, closing this issue.

closed 2023-08-31 by Rajashreekalmane