project-chip/connectedhomeip #27438: [Build] Darwin build failure

woody-apple, 2023-06-23:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5355597420/jobs/9714067297

Looks different than current ones.

### Platform

darwin

bzbarsky-apple, 2023-06-23:
The actual failure part of the log:
```
2023-06-23T11:52:03.9354410Z [1687521112101] [88373:222780] [DMG] Refresh LivenessCheckTime for 5224 milliseconds with SubscriptionId = 0xf507724e Peer = 02:0000000011223344
...

2023-06-23T11:52:03.9361730Z src/controller/tests/data_model/TestRead.cpp:1804: assertion failed: "callback.mOnError == 1"
2023-06-23T11:52:03.9361940Z src/controller/tests/data_model/TestRead.cpp:1805: assertion failed: "callback.mLastError == CHIP_ERROR_TIMEOUT"
2023-06-23T11:52:03.9362120Z src/controller/tests/data_model/TestRead.cpp:1806: assertion failed: "callback.mOnDone == 1"
2023-06-23T11:52:03.9362430Z '#3:','TestSubscribeAttributeTimeout                             ','FAILED'
2023-06-23T11:52:03.9362990Z [1687521117392] [88373:222780] [EM] <<< [E:12940i S:2 M:204948763] (S) Msg TX to 1:0000000012344321 [6EF6] --- Type 0001:03 (IM:SubscribeRequest)
```
Sadly, the failure bits don't come with Matter timestamps, just the harness timestamp, and that's much later than the Mater bits, afaict.

But `1687521117392 - 1687521112101 == 5291`, so it does look like we took longer than 5224ms to get to that next step... but it's still possible our timer here raced the liveness check timer.  We are currently adding 50ms in ComputeSubscriptionTimeout, but maybe we need to add more to avoid races....

woody-apple, 2023-06-23:
Another one here: https://github.com/project-chip/connectedhomeip/actions/runs/5360350319/jobs/9725105030

closed 2023-06-23 by mergify[bot]