project-chip/connectedhomeip #27439: [Build] Darwin build failure -[MTRControllerTests testControllerCATs]

woody-apple, 2023-06-23:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5357517058/jobs/9718483533

```
2023-06-23T16:01:38.0404340Z 2023-06-23 16:01:38.027649+0000 xctest[27542:294433] Write response: (
2023-06-23T16:01:38.0462230Z         {
2023-06-23T16:01:38.0463110Z         attributePath = "MTRAttributePath(100,200,300)";
2023-06-23T16:01:38.0463890Z     }
2023-06-23T16:01:38.0464550Z )
2023-06-23T16:01:38.0465650Z 2023-06-23 16:01:38.028318+0000 xctest[27542:299018] [Default] CHIP XPC connection disconnected
2023-06-23T16:01:38.0466790Z 2023-06-23 16:01:38.029031+0000 xctest[27542:294433] XPC connection disconnected
2023-06-23T16:01:38.0467930Z Test Case '-[MTRXPCProtocolTests testWriteAttributeSuccess]' passed (0.008 seconds).
2023-06-23T16:01:38.0469110Z Test Case '-[MTRXPCProtocolTests testXPCConnectionFailure]' started.
2023-06-23T16:01:38.0470210Z 2023-06-23 16:01:38.033018+0000 xctest[27542:294433] Device acquired. Reading...
2023-06-23T16:01:38.0471920Z 2023-06-23 16:01:38.033171+0000 xctest[27542:294433] [Default] Reading attribute ...
2023-06-23T16:01:38.0473730Z 2023-06-23 16:01:38.033374+0000 xctest[27542:300501] [Default] Cannot connect to XPC server for remote controller
2023-06-23T16:01:38.0474920Z 2023-06-23 16:01:38.033846+0000 xctest[27542:294433] [Default] Failed to obtain XPC connection
2023-06-23T16:01:38.0475980Z 2023-06-23 16:01:38.034157+0000 xctest[27542:294433] Read value: (null)
2023-06-23T16:01:38.0478360Z Test Case '-[MTRXPCProtocolTests testXPCConnectionFailure]' passed (0.004 seconds).
2023-06-23T16:01:38.0479510Z Test Suite 'MTRXPCProtocolTests' passed at 2023-06-23 16:01:38.036.
2023-06-23T16:01:38.0480410Z 	 Executed 27 tests, with 0 failures (0 unexpected) in 9.673 (9.701) seconds
2023-06-23T16:01:38.0481450Z Test Suite 'MatterTests.xctest' failed at 2023-06-23 16:01:38.037.
2023-06-23T16:01:38.0486350Z 	 Executed 256 tests, with 2 failures (0 unexpected) in 123.574 (123.846) seconds
2023-06-23T16:01:38.0487610Z Test Suite 'Selected tests' failed at 2023-06-23 16:01:38.038.
2023-06-23T16:01:38.0488520Z 	 Executed 256 tests, with 2 failures (0 unexpected) in 123.574 (123.852) seconds
2023-06-23T16:01:43.5116070Z 2023-06-23 16:01:43.510 xcodebuild[7950:251404] [MT] IDETestOperationsObserverDebug: 168.248 elapsed -- Testing started completed.
2023-06-23T16:01:43.5117230Z 2023-06-23 16:01:43.510 xcodebuild[7950:251404] [MT] IDETestOperationsObserverDebug: 0.000 sec, +0.000 sec -- start
2023-06-23T16:01:43.5118030Z 2023-06-23 16:01:43.511 xcodebuild[7950:251404] [MT] IDETestOperationsObserverDebug: 168.248 sec, +168.248 sec -- end
2023-06-23T16:01:48.1251490Z Failing tests:
2023-06-23T16:01:48.1251700Z 
2023-06-23T16:01:48.1254700Z 	-[MTRControllerTests testControllerCATs]
2023-06-23T16:01:48.1255720Z Test session results, code coverage, and logs:
2023-06-23T16:01:48.1256090Z 
2023-06-23T16:01:48.1256800Z 	/Users/runner/Library/Developer/Xcode/DerivedData/Matter-fezxjrcgbvmiwkgpcrhtmcinmudo/Logs/Test/Test-Matter Framework Tests-2023.06.23_15-48-40-+0000.xcresult
2023-06-23T16:01:48.1288860Z ** TEST FAILED **
2023-06-23T16:01:48.1289050Z 

```

bzbarsky-apple, 2023-06-23:
I've seen this a few times, but the log is completely useless in terms of what's going on...
```
2023-06-23T15:59:11.6374140Z Test Case '-[MTRControllerTests testControllerCATs]' started.
...
2023-06-23T15:59:11.9162680Z 2023-06-23 15:59:11.915706+0000 xctest[26431:291481] [Default] 4 CASE Authenticated Tags cannot be represented in a certificate.
...
2023-06-23T15:59:11.9233860Z 2023-06-23 15:59:11.922723+0000 xctest[26431:291481] [Default] 0x0 is not a valid CASE Authenticated Tag value.
2023-06-23T15:59:33.9411910Z xctest(27542,0x117b1a600) malloc: nano zone abandoned due to inability to preallocate reserved vm space.
2023-06-23T15:59:34.1884310Z Restarting after unexpected exit, crash, or test timeout in -[MTRControllerTests testControllerCATs]; summary will include totals from previous launches.
```

So we timed out, but why?  There's nothing async in this test, no waits.  The above logs correspond, afaict, to this part of the test:

```
    //
    // Trying to bring up the same fabric with too-long CATs should fail, if we
    // are taking the provided CATs into account.
    //
    params.nodeID = @(17);
    params.operationalKeypair = operationalKeys;
    params.caseAuthenticatedTags = tooLongCATs;
    controller = [factory createControllerOnExistingFabric:params error:nil];
    XCTAssertNil(controller);

    //
    // Trying to bring up the same fabric with invalid CATs should fail, if we
    // are taking the provided CATs into account.
    //
    params.nodeID = @(17);
    params.operationalKeypair = operationalKeys;
    params.caseAuthenticatedTags = invalidCATs;
    controller = [factory createControllerOnExistingFabric:params error:nil];
    XCTAssertNil(controller);

    [factory stopControllerFactory];
    XCTAssertFalse([factory isRunning]);
```

and we are in fact getting at least to that second `createControllerOnExistingFabric` call, which errors out with "0x0 is not a valid CASE Authenticated Tag value."....

What happened then?  Was this actually a crash?  If so what was the stack?  Log says nothing useful.

woody-apple, 2023-06-23:
I assume this is a real issue, and not just a test issue?   If we think it's a test issue, we should likely just disable it for now...

bzbarsky-apple, 2023-06-23:
I am assuming this is a real issue; I don't see anything in this test that would be weird in any way.

The problem is that there is no information on what the issue is in the log and I have been unable to reproduce it locally.

woody-apple, 2023-07-11:
Closing this, given I haven't seen this in a few days.

closed 2023-07-11 by woody-apple