project-chip/connectedhomeip #27445: [BUG] SetRegulatoryConfig command

jonsmirl, 2023-06-23:
### Reproduction steps

SetRegulatoryConfig is used to set in the radio regulatory country code.  How is it supposed to get copied into BasicInfo Attribute 6, Location? It currently isn't getting copied.

And then shouldn't BasicInfo Attribute 6, Location be read only because it is set via the SetRegulatoryConfig command? There shouldn't be two ways to set the radio regulatory region.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

42cf63b

### Platform

core

jonsmirl, 2023-06-23:
Issue report at Google.
https://issuetracker.google.com/issues/283470659

jonsmirl, 2023-06-23:
The certification tests should test to make sure that this regulatory region stuff is working correctly.  Minimum $25,000 FCC fine in USA if you get it wrong.

bzbarsky-apple, 2023-08-05:
> How is it supposed to get copied into BasicInfo Attribute 6, Location? 

It's copied right here: https://github.com/project-chip/connectedhomeip/blob/8363494ab658bdcdf652817c0657a9e71c3d7249/src/platform/DeviceControlServer.cpp#L52

> It currently isn't getting copied.

Where do you observe this?  We have CI tests that ensure that it is getting copied....

> And then shouldn't "BasicInfo Attribute 6, Location" be read only

I am guessing this got copied from Zigbee, and at this point the spec on this is not going to change.