project-chip/connectedhomeip #27449: [Build] Darwin test failed: -[MTRXPCListenerSampleTests test004_Subscribe]

woody-apple, 2023-06-24:
### Build issue(s)

https://pipelines.actions.githubusercontent.com/serviceHosts/24941cc8-9a45-4545-bbcd-b205b92ccb83/_apis/pipelines/1/runs/1870987/signedlogcontent/2?urlExpires=2023-06-24T19%3A53%3A59.8503687Z&urlSigningMethod=HMACV1&urlSignature=%2BKzIXRSSAVpiypMbLtNPYrZvU383NO0P%2BXyhUVzOmrU%3D

```
2023-06-24T18:21:50.6217360Z 2023-06-24 18:21:50.612434+0000 xctest[91898:228130] Device acquired. Deregistering...
2023-06-24T18:21:50.6217870Z 2023-06-24 18:21:50.612554+0000 xctest[91898:228130] [Default] Deregistering report handlers
2023-06-24T18:21:50.6218420Z 2023-06-24 18:21:50.612820+0000 xctest[91898:230517] [Default] CHIP XPC connection requests to stop reports
2023-06-24T18:21:50.6219100Z 2023-06-24 18:21:50.613456+0000 xctest[91898:228130] Deregistered
2023-06-24T18:21:50.6219670Z 2023-06-24 18:21:50.613513+0000 xctest[91898:230517] [Default] CHIP XPC connection disconnected
2023-06-24T18:21:50.6220160Z 2023-06-24 18:21:50.613873+0000 xctest[91898:228130] XPC connection disconnected
2023-06-24T18:21:50.6221030Z Test Case '-[MTRXPCProtocolTests testSubscribeAttributeWithParamsSuccess]' passed (0.010 seconds).
2023-06-24T18:21:50.6221880Z Test Case '-[MTRXPCProtocolTests testSubscribeClusterStateCacheFailure]' started.
2023-06-24T18:21:50.6222780Z 2023-06-24 18:21:50.617287+0000 xctest[91898:228130] [Default] Subscribing all attributes... Note that attributeReportHandler, eventReportHandler, and resubscriptionScheduled are not supported.
2023-06-24T18:21:50.6223440Z 2023-06-24 18:21:50.617596+0000 xctest[91898:230517] [Default] CHIP XPC connection established
2023-06-24T18:21:50.6223930Z 2023-06-24 18:21:50.618436+0000 xctest[91898:232603] XPC listener accepting connection
2023-06-24T18:21:50.6224450Z 2023-06-24 18:21:50.619196+0000 xctest[91898:228130] Subscribe attribute cache called
2023-06-24T18:21:50.6244200Z 2023-06-24 18:21:50.619999+0000 xctest[91898:232163] [Default] CHIP XPC connection disconnected
2023-06-24T18:21:50.6269130Z 2023-06-24 18:21:50.620309+0000 xctest[91898:228130] Report error received for attribute cache: Error Domain=MTRErrorDomain Code=1 "(null)"
2023-06-24T18:21:50.6284730Z 2023-06-24 18:21:50.620600+0000 xctest[91898:228130] XPC connection disconnected
2023-06-24T18:21:50.6286090Z 2023-06-24 18:21:50.620959+0000 xctest[91898:228130] Subscribe completion called with error: Error Domain=MTRErrorDomain Code=1 "(null)"
2023-06-24T18:21:50.6345500Z Test Case '-[MTRXPCProtocolTests testSubscribeClusterStateCacheFailure]' passed (0.007 seconds).
2023-06-24T18:21:50.6390620Z Test Case '-[MTRXPCProtocolTests testSubscribeClusterStateCacheSuccess]' started.
2023-06-24T18:21:50.6393550Z 2023-06-24 18:21:50.625085+0000 xctest[91898:228130] [Default] Subscribing all attributes... Note that attributeReportHandler, eventReportHandler, and resubscriptionScheduled are not supported.
2023-06-24T18:21:50.6394360Z 2023-06-24 18:21:50.625442+0000 xctest[91898:230517] [Default] CHIP XPC connection established
2023-06-24T18:21:50.6397650Z 2023-06-24 18:21:50.626043+0000 xctest[91898:230517] XPC listener accepting connection
2023-06-24T18:21:50.6443150Z 2023-06-24 18:21:50.626802+0000 xctest[91898:228130] Subscribe called
2023-06-24T18:21:50.6464730Z 2023-06-24 18:21:50.627358+0000 xctest[91898:228130] Attribute cache subscription succeeded for device 9876543210
2023-06-24T18:21:50.6489800Z 2023-06-24 18:21:50.627412+0000 xctest[91898:232163] [Default] CHIP XPC connection disconnected
2023-06-24T18:21:50.6491230Z 2023-06-24 18:21:50.627668+0000 xctest[91898:228130] XPC connection disconnected
2023-06-24T18:21:50.6492460Z 2023-06-24 18:21:50.627911+0000 xctest[91898:228130] Subscribe completion called with error: (null)
2023-06-24T18:21:50.6495640Z Test Case '-[MTRXPCProtocolTests testSubscribeClusterStateCacheSuccess]' passed (0.006 seconds).
2023-06-24T18:21:50.6530380Z Test Case '-[MTRXPCProtocolTests testSubscribeClusterStateCacheWithParamsSuccess]' started.
2023-06-24T18:21:50.6561330Z 2023-06-24 18:21:50.631113+0000 xctest[91898:228130] [Default] Subscribing all attributes... Note that attributeReportHandler, eventReportHandler, and resubscriptionScheduled are not supported.
2023-06-24T18:21:50.6591220Z 2023-06-24 18:21:50.631488+0000 xctest[91898:230517] [Default] CHIP XPC connection established
2023-06-24T18:21:50.6591790Z 2023-06-24 18:21:50.632185+0000 xctest[91898:232163] XPC listener accepting connection
2023-06-24T18:21:50.6592310Z 2023-06-24 18:21:50.632873+0000 xctest[91898:228130] Subscribe attribute cache called
2023-06-24T18:21:50.6592870Z 2023-06-24 18:21:50.633372+0000 xctest[91898:228130] Attribute cache subscription succeeded for device 9876543210
2023-06-24T18:21:50.6594000Z 2023-06-24 18:21:50.633422+0000 xctest[91898:230517] [Default] CHIP XPC connection disconnected
2023-06-24T18:21:50.6594590Z 2023-06-24 18:21:50.633668+0000 xctest[91898:228130] XPC connection disconnected
2023-06-24T18:21:50.6595110Z 2023-06-24 18:21:50.633914+0000 xctest[91898:228130] Subscribe completion called with error: (null)
2023-06-24T18:21:50.6595760Z Test Case '-[MTRXPCProtocolTests testSubscribeClusterStateCacheWithParamsSuccess]' passed (0.005 seconds).
2023-06-24T18:21:50.6596890Z Test Case '-[MTRXPCProtocolTests testSubscribeMultiAttributes]' started.
2023-06-24T18:21:50.6597500Z 2023-06-24 18:21:50.637197+0000 xctest[91898:228130] Device acquired. Subscribing...
2023-06-24T18:21:50.6598090Z 2023-06-24 18:21:50.637435+0000 xctest[91898:228130] [Default] Subscribing attribute ...
2023-06-24T18:21:50.6598620Z 2023-06-24 18:21:50.637698+0000 xctest[91898:232603] [Default] CHIP XPC connection established
2023-06-24T18:21:50.6599110Z 2023-06-24 18:21:50.637841+0000 xctest[91898:228130] [Default] Setup report handler
2023-06-24T18:21:50.6599610Z 2023-06-24 18:21:50.638255+0000 xctest[91898:232603] XPC listener accepting connection
2023-06-24T18:21:50.6600120Z 2023-06-24 18:21:50.639279+0000 xctest[91898:228130] [Default] Subscription established
2023-06-24T18:21:50.6692440Z 2023-06-24 18:21:50.640047+0000 xctest[91898:232603] [Default] Report received
2023-06-24T18:21:50.6693670Z 2023-06-24 18:21:50.640319+0000 xctest[91898:228130] Report value: (
2023-06-24T18:21:50.6694410Z         {
2023-06-24T18:21:50.6695590Z         attributePath = "MTRAttributePath(100,200,300)";
2023-06-24T18:21:50.6696270Z         data =         {
2023-06-24T18:21:50.6698050Z             type = SignedInteger;
2023-06-24T18:21:50.6761450Z             value = 123456;
2023-06-24T18:21:50.6762830Z         };
2023-06-24T18:21:50.6765610Z     }
2023-06-24T18:21:50.6767740Z )
2023-06-24T18:21:50.6790330Z 2023-06-24 18:21:50.641296+0000 xctest[91898:232163] [Default] Report received
2023-06-24T18:21:50.6834420Z 2023-06-24 18:21:50.641511+0000 xctest[91898:228130] Report value: (
2023-06-24T18:21:50.6835740Z         {
2023-06-24T18:21:50.6836190Z         attributePath = "MTRAttributePath(100,200,300)";
2023-06-24T18:21:50.6836500Z         data =         {
2023-06-24T18:21:50.6836750Z             type = SignedInteger;
2023-06-24T18:21:50.6837140Z             value = 771234;
2023-06-24T18:21:50.6837390Z         };
2023-06-24T18:21:50.6837600Z     }
2023-06-24T18:21:50.6837880Z )
2023-06-24T18:21:50.6838440Z 2023-06-24 18:21:50.641900+0000 xctest[91898:228130] Device acquired. Deregistering...
[... 135 lines cut ...]
2023-06-24T18:21:50.7099270Z Test Suite 'MatterTests.xctest' failed at 2023-06-24 18:21:50.702.
2023-06-24T18:21:50.7099640Z 	 Executed 256 tests, with 1 failure (0 unexpected) in 144.598 (144.851) seconds
2023-06-24T18:21:50.7100110Z Test Suite 'All tests' failed at 2023-06-24 18:21:50.703.
2023-06-24T18:21:50.7100480Z 	 Executed 256 tests, with 1 failure (0 unexpected) in 144.598 (144.856) seconds
2023-06-24T18:21:56.1514670Z 2023-06-24 18:21:56.151 xcodebuild[81415:204956] [MT] IDETestOperationsObserverDebug: 152.921 elapsed -- Testing started completed.
2023-06-24T18:21:56.1515520Z 2023-06-24 18:21:56.151 xcodebuild[81415:204956] [MT] IDETestOperationsObserverDebug: 0.000 sec, +0.000 sec -- start
2023-06-24T18:21:56.1516260Z 2023-06-24 18:21:56.151 xcodebuild[81415:204956] [MT] IDETestOperationsObserverDebug: 152.921 sec, +152.921 sec -- end
2023-06-24T18:22:00.2863110Z Failing tests:
2023-06-24T18:22:00.2863350Z 
2023-06-24T18:22:00.2864410Z 	-[MTRXPCListenerSampleTests test004_Subscribe]
2023-06-24T18:22:00.2864780Z Test session results, code coverage, and logs:
2023-06-24T18:22:00.2864990Z 
2023-06-24T18:22:00.2865090Z ** TEST FAILED **
[... cut ...]
2023-06-24T18:22:00.2866660Z 
2023-06-24T18:22:00.2896810Z 
2023-06-24T18:22:00.2897000Z Testing started
2023-06-24T18:22:05.4102120Z ##[error]Process completed with exit code 65.
2023-06-24T18:22:05.4207140Z ##[group]Run actions/upload-artifact@v3
2023-06-24T18:22:05.4207830Z with:
2023-06-24T18:22:05.4208100Z   name: darwin-framework-test-logs
2023-06-24T18:22:05.4209490Z   path: /tmp/darwin/framework-tests
2023-06-24T18:22:05.4209770Z   retention-days: 5
2023-06-24T18:22:05.4210030Z   if-no-files-found: warn
```

woody-apple, 2023-06-26:
Dang. I pasted the wrong link. Let me find.

woody-apple, 2023-06-26:
Found it @bzbarsky-apple https://github.com/project-chip/connectedhomeip/actions/runs/5366377183/jobs/9735769393

bzbarsky-apple, 2023-06-26:
So the relevant bit of that log, I think, is:

```
2023-06-24T23:37:33.2559580Z 2023-06-24 23:37:33.254109+0000 xctest[71751:186451] [Default] CHIP XPC connection disconnected
2023-06-24T23:37:33.2559840Z 2023-06-24 23:37:33.254125+0000 xctest[71751:187934] XPC connection disconnected
...
2023-06-24T23:37:35.2410360Z 2023-06-24 23:37:35.240229+0000 xctest[71751:186864] [DataManagement] ReportDataMessage =
...
2023-06-24T23:37:41.3413260Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/MTRXPCListenerSampleTests.m:777: error: -[MTRXPCListenerSampleTests test004_Subscribe] : Asynchronous wait failed: Exceeded timeout of 3 seconds, with unfulfilled expectations: "receive OnOff attribute report".
2023-06-24T23:37:41.3522950Z Test Case '-[MTRXPCListenerSampleTests test004_Subscribe]' failed (8.122 seconds).
```

So we got the data report, but never got that back to our actual completion.  I _think_ that's because the XPC connection disconnected, but not 100% sure.

I that's what it is, we've had these problems before; I thought the XPC machinery was keeping the XPC connection alive now...

bzbarsky-apple, 2023-06-26:
@kpark-apple @vivien-apple could you take a look, please?

kpark-apple, 2023-06-26:
Right. Somehow XPC connection was disconnected before it was supposed to be. Not sure how that's possible. Is there a way to see which commit SHA the test ran against?

bzbarsky-apple, 2023-06-26:
@kpark-apple It ran against a41398b648ee18024decd9c45f9ab3bdd3c7112f

kpark-apple, 2023-06-27:
I don't find any changes in https://github.com/project-chip/connectedhomeip/commit/a41398b648ee18024decd9c45f9ab3bdd3c7112f which might have affected the XPC connections.

bzbarsky-apple, 2023-07-07:
@kpark-apple Right, I expect this is some intermittent failure that is due to an existing issue we have....

woody-apple, 2023-07-20:
This appears to have been resolved

bzbarsky-apple, 2025-05-15:
Just happened again: https://github.com/project-chip/connectedhomeip/actions/runs/15026868875/job/42234231537

bzbarsky-apple, 2025-05-20:
Again: https://github.com/project-chip/connectedhomeip/actions/runs/15138745608/job/42558862117

andy31415, 2025-06-11:
Again: https://github.com/project-chip/connectedhomeip/actions/runs/15584352933/job/43888484018

```
2025-06-11T14:13:57.5756480Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/MTRXPCListenerSampleTests.m:813: error: -[MTRXPCListenerSampleTests test004_Subscribe] : Asynchronous wait failed: Exceeded timeout of 3 seconds, with unfulfilled expectations: "receive OnOff attribute report".
2025-06-11T14:13:57.5787200Z Test Case '-[MTRXPCListenerSampleTests test004_Subscribe]' failed (4.272 seconds).
```

andy31415, 2025-06-20:
note #37012 complains that some callbacks are not called (i.e. error handler is not called when subscribe fails). I wonder if we somehow have an intermittent failures in subscribe overall ... in both success and fail cases.