project-chip/connectedhomeip #27466: [Build] High memory consumption when compiling chip-tool.

ericzijian1994, 2023-06-26:
### Build issue(s)

1. GitHub hash: 7806dd19111513a61bb90d70a0e392dc89eeb454
2. Open a WSL2 terminal
3. Execute `scripts/examples/gn_build_example.sh examples/chip-tool out/chip-tool`
4. cpu and memory are above 90% and stuck at this step.
    ```sh
    + ninja -C out/chip-tool
    ninja: Entering directory `out/chip-tool'
    [0/18] ar libCHIP.a
    ```

woody-apple, 2023-06-26:
@ericzijian1994 Shouldn't this be filed against WSL2 if the memory isn't being reclaimed?

bzbarsky-apple, 2023-06-26:
@ericzijian1994 chip-tool has a fairly large binary, so large memory use during link is expected.

closed 2023-06-26 by bzbarsky-apple