project-chip/connectedhomeip #27473: CASE: Send Busy response when we are in the middle of handshake and receives sigma1

shubhamdp, 2023-06-26:
### Type

Other

### Explanation

Once https://github.com/project-chip/connectedhomeip/pull/27226 gets merged, the unsolicited handler for sigma1 will not be unregistered after receiving a sigma1, so we can send the busy response when we receives a sigma1 when we are in the middle of handshake.

### Platform

core (please add to version below)

### Testing

Manually tested with SDK

shubhamdp, 2023-07-06:
While looking into this I had three options... so though of putting up here.
1. Define a function and do similar stuff what [`SendStatusReport`](https://github.com/project-chip/connectedhomeip/blob/master/src/protocols/secure_channel/PairingSession.h#L135) from PairingSession.h is doing.
2. `SendStatusReport()` is not dependent on anything from PairingSession and it just requires exchange context which it takes as an argument, so can we make it static?
3. Move `SendStatusReport()` to ExchangeContext besides [`SendMessage`](https://github.com/project-chip/connectedhomeip/blob/master/src/messaging/ExchangeContext.cpp#L133) so that we call it using exchange context. And we stop duplicating Sending status response code.

Thoughts/

bzbarsky-apple, 2023-07-06:
@shubhamdp I really like option 3.  Presumably this will have SendStatusReport taking `const Protocols::SecureChannel::StatusReport &` or so?  Having that be a single utility on exchanges makes a _lot_ of sense.

bzbarsky-apple, 2023-07-06:
I guess another option would be to have a `SendOnExchange` on `SecureChannel::StatusReport`, but that seems like a bit of a layering issue.....

shubhamdp, 2023-07-21:
1. Going with this... and implemented in the PR #28153
2. Since SendStatusReport is protected and I did not feel good about making it private, so avoided that path.
3. I see the cyclic dependency issue so, could not get around it

I'll raise an issue to see if I can optimize the SendStatusReport bits to avoid code duplication.

bzbarsky-apple, 2023-07-21:
> I see the cyclic dependency issue so, could not get around it

@shubhamdp what was the cyclic dependency?

shubhamdp, 2023-07-26:
After https://github.com/project-chip/connectedhomeip/pull/28219 got merged, going with option 3, where we add `SendStatusReport()` API on `ExchangeContext`.

PR #28153 (2023-08-03): CASE: Send busy status report if we receive a sigma1 and we are in the middle of handshake
With #27226,  Unsolicited message handler will never be unregistered and CASEServer can still receive a Sigma1. If we are in the middle of a handshake then do send out Busy status report.

Fixes #27473

#### Tests
- Tested with lighting-app/esp32 and chip-tool. [lightin-app/linux was too quick to reproduce this behaviour]
```
# Commissioning and access control
./chip-tool pairing onnetwork-long 21 20202021 3840
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode":2, "subjec [... cut]
closed 2023-08-03 by mergify[bot]