project-chip/connectedhomeip #27478: [Build] Test suite on linux failed

woody-apple, 2023-06-26:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5376979026/jobs/9754861432?pr=27472

```
ERROR   12:08:59.759 - CHIP_REPL_YAML_TESTER OUT: [1687781339.759214][41516:41526] CHIP:CTL: Unknown filter type; all matches will fail
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=1
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with _matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with 387FE5E77EFE6E10._matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with 6EA6A637A5520000.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with 6EA6A637A5520000.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with _V65521._sub._matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with _T65535._sub._matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with _S15._sub._matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with _L3840._sub._matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with _CM._sub._matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: Responding with 387FE5E77EFE6E10._matterc._udp.local
ERROR   12:08:59.759 - APP  OUT  : CHIP:DIS: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 387FE5E77EFE6E10.
ERROR   12:08:59.760 - CHIP_REPL_YAML_TESTER OUT: [1687781339.760566][41516:41526] CHIP:CTL: Unknown filter type; all matches will fail
ERROR   12:08:59.761 - APP  OUT  : CHIP:DIS: mDNS service published: _matterc._udp
ERROR   12:08:59.761 - CHIP_REPL_YAML_TESTER OUT: [1687781339.761332][41516:41526] CHIP:CTL: Unknown filter type; all matches will fail
ERROR   12:09:05.694 - CHIP_REPL_YAML_TESTER ERR: WARNING:root:Test step failure in Wait for the commissioned device to be retrieved
ERROR   12:09:05.694 - CHIP_REPL_YAML_TESTER ERR: WARNING:root:PostProcessCheckStatus.ERROR: The test expects no error but the "FAILURE" error occured.
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT: Traceback (most recent call last):
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:   File "/__w/connectedhomeip/connectedhomeip/scripts/tests/chiptest/yamltest_with_chip_repl_tester.py", line 146, in main
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:     asyncio.run(execute_test(yaml, runner))
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:   File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:     return loop.run_until_complete(main)
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:   File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:     return future.result()
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:   File "/__w/connectedhomeip/connectedhomeip/scripts/tests/chiptest/yamltest_with_chip_repl_tester.py", line 84, in execute_test
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT:     raise Exception(f'Test step failed {test_step.label}')
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT: Exception: Test step failed Wait for the commissioned device to be retrieved
ERROR   12:09:05.697 - CHIP_REPL_YAML_TESTER OUT: 
```

woody-apple, 2023-06-26:
@andy31415, who is the best to handle the REPL stuff these days?

andy31415, 2023-06-26:
I do not think we have dedicated repl test execution resources currently. Depending on repetability/recurrence of this issue we likely will need to find a volunteer from our general maintainer pool.

woody-apple, 2023-06-26:
I've seen it twice in a day, so it's repeatable.

andy31415, 2023-10-10:
I have not seen this in a while now, marking as closed/obsolete

closed 2023-10-10 by andy31415