project-chip/connectedhomeip #27479: [Build] pyyaml failed

woody-apple, 2023-06-26:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5376188056/jobs/9753037388?pr=27470

```
[785/795] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../scripts/py_matter_yamltests --default-toolchain=//build/toolchain/host:mac_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py.pw_pystamp --capture-output --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py_metadata_path_list.txt -- ../../scripts/py_matter_yamltests/test_yaml_loader.py
[786/795] touch python/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py.stamp
[787/795] touch obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py.stamp
[788/795] touch obj/scripts/py_matter_yamltests/matter_yamltests.tests.stamp
[789/795] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../src/transport/tests --default-toolchain=//build/toolchain/host:mac_x64_gcc --current-toolchain=//build/toolchain/host:mac_x64_gcc --touch gen/src/transport/tests/TestSessionManager_run.pw_pystamp --capture-output --module pw_unit_test.test_runner --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files gen/src/transport/tests/TestSessionManager_run_metadata_path_list.txt -- --runner ../../third_party/pigweed/repo/targets/host/run_test --test tests/TestSessionManager
[790/795] touch obj/src/transport/tests/TestSessionManager_run.stamp
[791/795] touch obj/src/transport/tests/tests_run.stamp
ninja: build stopped: cannot make progress due to previous errors.
Error: Process completed with exit code 1.
```

bzbarsky-apple, 2023-06-26:
```
src/messaging/tests/TestExchangeMgr.cpp:189: assertion failed: "!sendDelegate.IsOnResponseTimeoutCalled"
```

this has nothing to do with YAML or python.

bzbarsky-apple, 2023-06-26:
I am guessing we lost timeslice for > 100ms, so the timeout actually fired....

woody-apple, 2023-06-26:
Thanks, renaming!

bzbarsky-apple, 2023-06-26:
In particular:
```
[1687774193119] [15821:279924] [IN] SecureSession[0x600002e340f0]: Allocated for Test Type:1 LSID:2
'#3:','Test OnConnectionExpired timeout handling ','PASSED'
[1687774193258] [15821:279924] [EM] <<< [E:47080i S:2 M:40781666] (S) Msg TX to 1:0000000012344321 [6EF6] --- Type 0002:01 (BDX:SendInit)
[1687774193259] [15821:279924] [IN] (S) Sending msg 40781666 on secure session with LSID: 2
```

Note the 139ms time gap between previous test completion and us logging our send.... That's us losing the timeslice.

There's really no way to win here.  We could bump the exchange timeout value to make this less likely, but of course then the test will always run slower.... and nothing guarantees that we won't lose the timeslice for a longer time either.

woody-apple, 2023-07-07:
@bzbarsky-apple Let's up the timeout significantly here, it won't slow the test down that much compared to overall runtime.

closed 2023-07-11 by mergify[bot]