project-chip/connectedhomeip #27490: [Build] OpenIOT Build failure

woody-apple, 2023-06-26:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5382843021/jobs/9768812925

```
=========================== short test summary info ============================
FAILED src/test_driver/openiotsdk/integration-tests/lock-app/test_app.py::test_smoke_test - AssertionError: FVPdev: Didn't find Open IoT SDK lock-app example application start in 10 s
FAILED src/test_driver/openiotsdk/integration-tests/lock-app/test_app.py::test_commissioning - AssertionError: FVPdev: Didn't find Open IoT SDK lock-app example application start in 10 s
FAILED src/test_driver/openiotsdk/integration-tests/lock-app/test_app.py::test_lock_ctrl - assert (None is not None)
======================== 3 failed, 3 warnings in 33.82s ========================
```

### Platform

other

woody-apple, 2023-06-26:
ping @ATmobica

ATmobica, 2023-06-27:
It looks like we use a super low-performance CI runner for this job. The FVP application starts but is very slow. We need to keep in mind that FVP is a simulator so it depends heavily on CPU performance. I'll increase the default timeout for waiting for a specific output. This should prevent such situations.

closed 2023-06-29 by ATmobica