project-chip/connectedhomeip #27492: [BUG]

nicelyjust, 2023-06-27:
### Reproduction steps

SOE：moto e5 Android 8.0
ABI：armeabi-v7a
1. read readDescriptorClusterDeviceListAttribute by chip-tool(Android)
2. then get a wrong and strange devicType

problem log 
`getDeviceType: DescriptorClusterDeviceTypeStruct {
                                                                                                    	deviceType: -8887494820303911500
                                                                                                    	revision: 1
                                                                                                    }`

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master latest

### Platform

android

### Anything else?

It is likely to be a problem with the ABI platform

bzbarsky-apple, 2023-06-27:
@andy31415 @yunhanw-google

andy31415, 2023-06-28:
@nicelyjust could you provide reproduction details?

Even after reading the summary, I do not actually know what is being run and how. It references chip-tool (but unsure if UI version or binary compilation) and it does not describe what application it connects to (e.g. is it some light example app or all clusters or something else?)

I would need: what apps are running (there should be 2: one device, one controller) how they are setup and what is the command run and what you get instead of expected. If you have some investigation (like chip-tool on linux x64 works, but not on something else) that would be also great.

buddman1208, 2023-07-05:
@andy31415 @bzbarsky-apple 
Dear developers, We have a similar issue. 
If you don't mind, could you check my issue link?
Thank you.
https://github.com/project-chip/connectedhomeip/issues/27621

yunhanw-google, 2024-01-14:
@buddman1208  @nicelyjust  this should be resolved in https://github.com/project-chip/zap/pull/1092/files, please reopen if the issue is still there, thanks

closed 2024-01-14 by yunhanw-google