project-chip/connectedhomeip #27511: [Documentation] Simple Linux controller example

samsta, 2023-06-27:
### Documentation issues

It would be helpful to add a simple C++ example app that shows how to control a (previously commissioned) device _without_ using `chip-tool`, e.g.

- read an attribute
- issue a command
- subscribe to attribute changes

`chip-tool` is great for experimenting and exploring, but not very useful to build an application on top. If such an example exists already, it would be good to make this more obvious as I haven't found it yet.

Thanks for considering this request,
Sam

### Platform

other

MatheoDumont, 2023-07-27:
So Samsta, did you manage to find informations or documentation to help you ? I am looking for infos as well.

samsta, 2023-07-29:
No, I ended up using parts of chip-tool and based my implementation on that

topraveenpaul, 2023-11-07:
> No, I ended up using parts of chip-tool and based my implementation on that
> […](#)

@samsta  Would you mind sharing  your code implementation that uses parts of chip-tool? I'm on the same boat.

samsta, 2023-11-07:
Sure, @topraveenpaul . I have attached a library with the controller and some helper classes, and an example client implementing an OnOff device using the controller. I have replaced the bits that are company specific with more generic code but I haven't tried to compile after the change, so you're on your own here. But it should give you a starting point.

This is built out-of-tree, so the lib assumes that you've built and staged the artifacts from connectedhomeip appropriately - not a work flow supported by this project out of the box (unfortunately) but it's the way we're doing it.

[matter-app-lib.zip](https://github.com/project-chip/connectedhomeip/files/13285772/matter-app-lib.zip)
[OnOffChipLib.zip](https://github.com/project-chip/connectedhomeip/files/13285885/OnOffChipLib.zip)

topraveenpaul, 2023-11-07:
> Sure, @topraveenpaul . I have attached a library with the controller and some helper classes, and an example client implementing an OnOff device using the controller. I have replaced the bits that [...]

Many thanks @samsta . I think this would really help.