project-chip/connectedhomeip #27517: [BUG] chip-tool test will fail if the expected number of events is inconsistent

ericzijian1994, 2023-06-28:
### Reproduction steps

1. The Test_TC_SMCO_2_2.yaml file I added expects an event.
    ```sh
    tests:
        - label: "TH reads AllClear event from DUT"
          PICS: SMCO.S.E00010
          command: "readEvent"
          event: "AllClear"
          response:
              - values:
                    - value: {}
    ```

2. Three cases with different number of events.
    1. The case of empty events.
        [readEvent_log_0.txt](https:github.com/project-chip/connectedhomeip/files/11846788/readEvent_log_0.txt)
        Get `Test Failure: Test_TC_SMCO_2_2`

    2. The case of one event.
        [readEvent_log_1.txt](https:github.com/project-chip/connectedhomeip/files/11846790/readEvent_log_1.txt)

    3. The case of two events.
        [readEvent_log_2.txt](https:github.com/project-chip/connectedhomeip/files/11846791/readEvent_log_2.txt)
        Get `Test Failure: zzz_generated/chip-tool/zap-generated/test/Commands.h:57604: CHIP Error 0x0000002F: Invalid argument`

3. If TH tests, when an error is encountered, it will enter a dead loop.
[image]

By the way, is there a way to clear `EventReportIBs`?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

33437606053d7b0da04ee0b546603de97f95e6af

### Platform

other

bzbarsky-apple, 2023-06-28:
Failing is correct here, if the server sends a number of events different from 1, given the above YAML: the test says "there should be one event here", so if there is a different number of events the test should fail.

If that's not what you want to be testing, you should have different YAML.

> By the way, is there a way to clear EventReportIBs?

Not a standard one.  Specific devices might have device-specific things.

>  when an error is encountered, it will enter a dead loop

That sounds like a bug in the TH, which should be filed in https://github.com/CHIP-Specifications/chip-certification-tool, not here.

closed 2023-06-28 by bzbarsky-apple

ericzijian1994, 2023-06-29:
> That sounds like a bug in the TH, which should be filed in https://github.com/CHIP-Specifications/chip-certification-tool, not here.

@bzbarsky-apple If `chip-tool` adds `--continueOnFailure`, it will always return an error, which is not a problem for TH.

bzbarsky-apple, 2023-07-07:
> If chip-tool adds --continueOnFailure

Please file an issue with steps to reproduce, including the _exact_ command that was executed (notably missing from test_log_3.txt).