project-chip/connectedhomeip #27527: For full nocChain how to configure cat for admin auth

robinmo, 2023-06-28:
Found that controller startup params init method have 2 way. For full noc chain init method it has noc icac rcac params but no cats , and then it goes into local noc chain generation the cat is kUndefined value. How to configure cat?
 
I had tried init with the full chain way and set noc chain issuer for controller but failed to find operational.

robinmo, 2023-06-28:
@aajain-com @Bhaskar-Sarma

robinmo, 2023-06-28:
@bzbarsky-apple could u help or have any suggestion?

bzbarsky-apple, 2023-06-29:
@robinmo When generating your NOC for the "full noc chain" method, give that NOC the CATs you want.

There should be no "local NOC chain generation" going on if you use that init method.

closed 2023-06-29 by bzbarsky-apple

robinmo, 2023-06-29:
> @robinmo When generating your NOC for the "full noc chain" method, give that NOC the CATs you want.
> 
> There should be no "local NOC chain generation" going on if you use that init method.

The problem is that I use our app as the controller and need adding another hub to have adaut right. Then I initWithIPK... for startupParams but how to set the cats for controller?

robinmo, 2023-06-29:
@bzbarsky-apple PS: In IOS 16.4 Matter SDK startupParams has no caseAuthTags property

bzbarsky-apple, 2023-07-08:
@robinmo If you ar doing "full nocChain", why do you care what's in startupParams?

> Then I initWithIPK... for startupParams but how to set the cats for controller?

By generating a NOC with those CATs and doing the "initWithIPK" version that takes a NOC, no?

robinmo, 2023-07-18:
The matter is that I would set another endpoint as admin throughout CAT，No
startupParams no cat configure field.

bzbarsky-apple, 2023-07-20:
> The matter is that I would set another endpoint as admin throughout CAT

What does "another endpoint as admin" mean?   CATs are attached to nodes, not endpoints....

bzbarsky-apple, 2023-07-21:
OK, so you can use the "here are the certs" API to set up the controller on that hub...

robinmo, 2023-08-04:
@bzbarsky-apple
Hi, me, again😅. 

Need your help.
Now I use iOS 16.4 Matter SDK's MatterSupport to commission a new matter iot device, our app act as an controller. 

MatterSupport commission success, but during our fabric commissioning, always failed in FindOperational step.
--
Reproduce：
Xcode 14.3, iPhone (any > 16.4) one app as controller
Preset: iPhone connect to the hub's(maybe a wifi router) main network, the commission the matter iot device into iot network.(iot network is not isolated from main network)

1.Use MatterSupport to commission into apple fabric
2.Use matter SDK throughout full NOC Chain to commission into another third-party fabric.
--

Details relative log
```
Aug  4 17:53:43 iPhone8-tsukePrivate Deco(Matter)[1413] <Notice>: Successfully finished commissioning step 'SendNOC'
Aug  4 17:53:43 iPhone8-tsukePrivate Deco(Matter)[1413] <Notice>: Performing next commissioning step 'FindOperational'
Aug  4 17:53:44 iPhone8-tsukePrivate Deco(Matter)[1413] <Notice>: Error on commissioning step 'FindOperational': '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1723: CHIP Error 0x00000054: Invalid CASE parameter'
Aug  4 17:53:44 iPhone8-tsukePrivate Deco(Matter)[1413] <Error>: Failed to perform commissioning step 20
Aug  4 17:53:44 iPhone8-tsukePrivate Deco(Matter)[1413] <Notice>: Going from commissioning step 'FindOperational' with lastErr = '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1723: CHIP Error 0x00000054: Invalid CASE parameter' -> 'Cleanup'
Aug  4 17:53:44 iPhone8-tsukePrivate Deco(Matter)[1413] <Notice>: Performing next commissioning step 'Cleanup' with completion status = '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1723: CHIP Error 0x00000054: Invalid CASE parameter'
Aug  4 17:53:44 iPhone8-tsukePrivate Deco(Matter)[1413] <Notice>: Successfully finished commissioning step 'Cleanup'
```

bzbarsky-apple, 2023-08-05:
Well, the other side responded with "invalid CASE parameter".

Can you commission one of the test Matter apps with your app?  That would give you server-side logs to work with and might tell you what the server does not like.  But chances are, the way you set up your operational certificates and key for your client is wrong somehow.  Hard to tell how without knowing exactly what you did.

robinmo, 2023-08-10:
Finally, found that the fabric wireless encryption type is WPA3. With none or WPA/WPA2, success.
Thanks anyway.
I appreciate your every time quick help，thank you so much for adapting to matter for our app.
Yesterday, we have make it stable for matter commissioning with our full chain certs.:tada:

bzbarsky-apple, 2023-08-10:
@robinmo Congratulations!

That said, I am quite curious why WPA3 vs WPA2 would affect the CASE handshake...

robinmo, 2023-08-14:
> @robinmo Congratulations!
> 
> That said, I am quite curious why WPA3 vs WPA2 would affect the CASE handshake...

So many iot device in the market haven't yet supported joining WPA3 encryption wireless.

qawitherev, 2023-08-22:
> @bzbarsky-apple Hi, me, again😅.
> 
> Need your help. Now I use iOS 16.4 Matter SDK's MatterSupport to commission a new matter iot device, our app act as an controller.
> 
> ## MatterSupport commission success, but during our fabric commissioning, always failed in FindOperational step.

hi @robinmo i am trying to use the MatterSupport to bring my matter device onto a local network and commission it onto a fabric. but it seems that the requestHandler class (inside the matter extension) didn't get executed thus commissionDevice() to commissionDevice did not run. Is the commissionDevice() need to be called somewhere or is called automatically when MatterSupport done doing its work because the latter seems is not happening in my case. How is your commissionDevice() called in your code?

robinmo, 2023-08-24:
> hi @robinmo i am trying to use the MatterSupport to bring my matter device onto a local network and commission it onto a fabric. but it seems that the requestHandler class (inside the matter [...]

Hi, @qawitherev , do you mean handler's commissionDevice() below?
[image]

do you follow the guide: https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem#Enable-Developer-Mode

Have u configured MatterSupport payload for singing & capabilities?
[image]

qawitherev, 2023-08-24:
> Hi, @qawitherev , do you mean handler's commissionDevice() below? [image]

@robinmo 
Yep. Turns out we have to use logger to log extensions. I am now able to commission my device using matter framework after bringing it into a local network using matterSupport. 👨🏼‍🦯

Do we have to include the matter payload setup capabilities? I just included it in but not really sure what it does.

robinmo, 2023-08-24:
> @robinmo Yep. Turns out we have to use logger to log extensions. I am now able to commission my device using matter framework after bringing it into a local network using matterSupport. 👨🏼‍🦯
>
> Do we [...]

https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_developer_matter_allow-setup-payload/

qawitherev, 2023-08-24:
> https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_developer_matter_allow-setup-payload/

I see... Thanks!

robinmo, 2023-08-24:
@qawitherev my pleasure. Have your team use full cert chain to commission or just the common way that assigned noc by the framework?

qawitherev, 2023-08-24:
@robinmo 

Yeah about that, haven't really dig much about the certification thing. I just use what [CHIPTool](https://github.com/project-chip/connectedhomeip/tree/master/src/darwin/CHIPTool)'s fabric key for the key pair thing(?) since I am just trying to build prototype first using the provided ESP board provided by my firmware guy.

Do you have any refs on resources I can read regarding this thing? I've read the commission process on Google Dev and Nordic web, but haven't really have a firm grasp on it yet.

robinmo, 2023-08-24:
you can read the [Matter 1.0 Core Specification](https://csa-iot.org/developer-resource/specifications-download-request/)
[image]
the chip is the implementation of this specification.
If you do have problem about making commission, just share and let me know if i can help.

qawitherev, 2023-08-24:
@robinmo 
alright, thanks!