project-chip/connectedhomeip #27539: [Feature] Get the latest eventNumber in the chip-tool test

ericzijian1994, 2023-06-29:
### Feature description

There is a possible solution to get the event number so that I can read the specified event, but it requires adding the Unit Testing cluster.

```sh
- label: "Generate an event on the accessory"
  cluster: "Unit Testing"
  endpoint: 0
  command: "TestEmitTestEventRequest"
  arguments:
      values:
          - name: "arg1"
            value: 1
          - name: "arg2"
            value: 2
          - name: "arg3"
            value: true
  response:
      values:
          - name: "value"
            saveAs: eventNumber
```

Is it possible to add a simpler command to get the latest eventNumber?

### Platform

other

bzbarsky-apple, 2023-07-06:
So this is asking for a chip-tool command that will do a wildcard read on events on the device and then return the largest observed event number?  @ericzijian1994

ericzijian1994, 2023-07-07:
> So this is asking for a chip-tool/YAML command that will do a wildcard read on events on the device and then return the largest observed event number? @ericzijian1994

Yes.

andy31415, 2023-08-25:
Re-open: need to make sure we have this functionality runnable in yaml tests.

PR #28704 (2023-08-24): Added method in python runner to get eventNumber for TC-SMOKECO tests
Proposing the method to fixes #27539

- This PR tried to -

Adding a method in Python runner to get eventNumber in order to get the TC-SMOKECO test to work, I understand this could not be a good implementation and may need improvement/use another alternative. 

- The problem was -

The `readEvent` command from eventTest returns a list of events. But in yaml, the value I expect needs to match the value actually returned. And the eventNumber parameter in yaml acts as a filter that allows me to accomplish this. 
If I know the e [... cut]