project-chip/connectedhomeip #27542: [BUG] Did matter protocol support Android5.0 and 6.0?

luobosu, 2023-06-29:
### Reproduction steps

I change the minSDKVersion of the Android chip-tool app to 21 ，but it will crash on Android 5.0 device ：

`

2023-06-29 14:27:02.291 24427-24427 SetupPayloadParser      com.google.chip.chiptool             E  Cannot load library.
                                                                                                    java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "__register_atfork" referenced by "libSetupPayloadParser.so"...
                                                                                                    	at java.lang.Runtime.loadLibrary(Runtime.java:371)
                                                                                                    	at java.lang.System.loadLibrary(System.java:989)
                                                                                                    	at chip.setuppayload.SetupPayloadParser.<clinit>(SetupPayloadParser.java:78)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.handleScannedQrCode$lambda-10(BarcodeFragment.kt:199)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.lambda$KQ_xOxz0LBqlN0rG11wbU9zHO7E(BarcodeFragment.kt)
                                                                                                    	at com.google.chip.chiptool.setuppayloadscanner.-$$Lambda$BarcodeFragment$KQ_xOxz0LBqlN0rG11wbU9zHO7E.run(lambda)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:739)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                                    	at android.os.Looper.loop(Looper.java:145)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:5942)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at java.lang.reflect.Method.invoke(Method.java:372)
                                                                                                    	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
`

It seems that __register_atfork can not be used on Android 5.0&6.0 .  
Does matter protocol support android 5.0 & 6.0 ?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

TAG 1.0.0.2

### Platform

android

### Anything else?

.

yunhanw-google, 2023-08-01:
@luobosu per https://android-review.googlesource.com/c/platform/bionic/+/948880/1/android-changes-for-ndk-developers.md, "__register_atfork" need android 23, so it means at least android 6.

Recently we have added pure kotlin SetupPayloadParser, @yufengwangca , @luobosu maybe this issue has disappeared, please retry? thanks

andy31415, 2023-08-01:
The requirement as far as I am aware is API level 26 at a minimu, so you actually need android 8 and above. Reason is IPv6 support requirements.

andy31415, 2023-08-01:
Just changing "minSDKVersion" is not sufficient for making chip work on old android versions. They will generally not work because of ipv6 support needs and the SDK does not support such a setup. Use android 8 or above at least.

closed 2023-08-01 by andy31415