project-chip/connectedhomeip #27570: Add `LastChangedTime` to Resource Monitoring Cluster Definition

p0fi, 2023-06-30:
# 📖 Description
The discussion around the replacement product list caused spec changes. It added 2 new attributes. This issue should only add the `LastChangedTime` to the cluster definition since its unclear if the second attribute will make it into to spec in time.

robszewczyk, 2023-06-30:
Possible dup of https://github.com/project-chip/connectedhomeip/issues/27562. If tracking this one for ongoing work / resolution, please note this issue number in the spec PR

p0fi, 2023-06-30:
This is no straight forward dup. It's by design split up into more than one issue (see #27571) to track the work and help make the PRs the right scope and size. Although I was not aware of https://github.com/project-chip/connectedhomeip/issues/27562 existing :)

PR #27603 (2023-07-06): Add `LastChangedTime` to Resource Monitoring Cluster Definition
# 📖 Description
This PR adds the `LastChangedTime` Attribute to the [resource-monitoring-cluster.xml](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/resource-monitoring-cluster.xml)

The `ReplacementProductList` will be handled by this issue https://github.com/project-chip/connectedhomeip/issues/27571

## 🔎 Details
- Code generation ran with `zap_regen_all.py`

## 🧪 How Has This Been Tested
- chip-tool was built using `./scripts/examples/gn_build_example.sh examples/chip-tool  [... cut]
closed 2023-07-06 by mergify[bot]