project-chip/connectedhomeip #27575: [1.2] Time sync - tracking

cecille, 2023-06-30:
### Reproduction steps

PR inspection

### Bug prevalence

always

### GitHub hash of the SDK that was being used

abb7a802231f2b615b3030c5df43f5144871e09c

### Platform

core

### Type

Common Cluster Logic

### Anything else?

- [ ]  Figure out what happens if the TZ offset is out of range AND the code fails to load a time zone. See https://github.com/project-chip/connectedhomeip/pull/26082#discussion_r1240951613. In particular, which error code needs to be returned (I think this still needs to be a constraint error), AND can we do this check in a way that prevents making changes that necessitate reloading the time
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082#discussion_r1240951626 - align errors between the ember command parser and the
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1212444174 - better overflow protection when going from chip epoch to unix epoch
- [ ] handlers for 3.5.6.1 SHOULD - If a Node has used the Last Known Good UTC Time, it SHOULD recheck its security materials and existing connections if it later achieves time synchronization.
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242621779 - time zone name copying
- [ ] document event generation on time zone name only change. https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242626060
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242627777 - change event flag from parameter to return
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242630331 - change size comparison
- [ ] follow up on ScheduleDelayedAction function https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242638706
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242689236 - adjust TZ / DST list on time change
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242639975 - propagate error for set or document why the error changes
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1241153735 - tighten up time get / set features to move them closer to the receive / send boundary
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242642964 - remove casts on TZ / DST offset
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242655854 - refactor DST list parsing code
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242657708 - remove mEventFlag
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242658973 - minor code suggestion from Boris
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242660231 - don't encode local time on error
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242667611 , https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242668186 use unknown value check for enums
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242673209 - double check error code for invalid enum
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242671601 - tighten up granularity check
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242676471 - fix how we're setting granularity
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1242683543 - move event generation into the instance rather than handling in the ember functions
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082/files#r1240968943 - fix time source check to allow domain name addresses
- [ ] https://github.com/project-chip/connectedhomeip/pull/26082#discussion_r1242685200 - move the IP / domain address checks out of the delegate - this is not something the app needs to care about.
- [ ]

cecille, 2023-11-08:
Woah - why was this closed?

fessehaeve, 2023-11-09:
Thanks @cecille for the catch. I have started working on some of these already.