project-chip/connectedhomeip #27579: [Build] Darwin tests failing quite a bit

woody-apple, 2023-06-30:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5423925004/jobs/9862553111

```
ERROR   15:57:37.999 - PAIR OUT  : [IN] (U) Sending msg 185534853 to IP address 'UDP:[::1%lo0]:5640'
ERROR   15:57:37.999 - PAIR OUT  : [EM] Flushed pending ack for MessageCounter:184811596 on exchange 38632i
ERROR   15:57:37.999 - PAIR OUT  : [EM] >>> [E:38632i S:0 M:184811597 (Ack:185534852)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
ERROR   15:57:37.999 - PAIR OUT  : [EM] OnMessageReceived failed, err = src/messaging/ExchangeMgr.cpp:305: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
ERROR   15:57:37.999 - PAIR OUT  : [EM] >>> [E:38632i S:0 M:184811598 (Ack:185534852)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
ERROR   15:57:37.999 - PAIR OUT  : [EM] OnMessageReceived failed, err = src/messaging/ExchangeMgr.cpp:305: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
ERROR   15:57:37.999 - PAIR OUT  : [EM] >>> [E:38632i S:0 M:184811599 (Ack:185534852)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
ERROR   15:57:37.999 - PAIR OUT  : [EM] OnMessageReceived failed, err = src/messaging/ExchangeMgr.cpp:305: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
ERROR   15:57:37.999 - PAIR OUT  : [EM] >>> [E:38632i S:0 M:184811600 (Ack:185534852)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
ERROR   15:57:37.999 - PAIR OUT  : [EM] OnMessageReceived failed, err = src/messaging/ExchangeMgr.cpp:305: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
ERROR   15:57:37.999 - PAIR OUT  : [CTL] Discovery timed out
ERROR   15:57:38.000 - PAIR OUT  : [CTL] Stopping commissioning discovery over DNS-SD
ERROR   15:57:38.000 - PAIR OUT  : [DIS] Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)
ERROR   15:57:38.000 - PAIR OUT  : [TOO] Secure Pairing Failed
ERROR   15:57:38.000 - PAIR OUT  : [TOO] Run command failure: ../../examples/chip-tool/commands/pairing/PairingCommand.cpp:294: CHIP Error 0x00000003: Incorrect state
ERROR   ================ CAPTURED LOG END ====================
ERROR   Test_TC_KEYPADINPUT_3_3        - FAILED in 34.90 seconds

```

### Anything else?

Been happening quite a bit the last couple days, don't see an obvious PR that could have caused this.

woody-apple, 2023-06-30:
This is happening pretty often

bzbarsky-apple, 2023-07-06:
Relevant part of the log looks like this, interleaving the things that should be interleaved (note that the timestamps are wrong; they have to do with when we dumped the log files, not when things happened... I am ordering things based on the actual exchange and message ids involved; I wish we were including timestamps in our actual logging here):
```
ERROR   15:57:37.998 - PAIR OUT  : [EM] <<< [E:38632i S:0 M:185534852] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
...
ERROR   15:57:16.115 - APP  OUT  : [1688140636112] [12778:76445] [EM] >>> [E:38632r S:0 M:185534852] (U) Msg RX from 0:7B06C856CC92BCB5 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
...
ERROR   15:57:16.115 - APP  OUT  : [1688140636113] [12778:76445] [EM] <<< [E:38632r S:0 M:184811596 (Ack:185534852)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
ERROR   15:57:16.115 - APP  OUT  : [1688140636114] [12778:76445] [IN] (U) Sending msg 184811596 to IP address 'UDP:[::1%lo0]:51818'
...
ERROR   15:57:37.998 - PAIR OUT  : [EM] Retransmitting MessageCounter:185534852 on exchange 38632i Send Cnt 1
...
ERROR   15:57:16.115 - APP  OUT  : [1688140636114] [12778:76445] [IN] Received a duplicate message with MessageCounter:185534852 on exchange 38632r
ERROR   15:57:16.118 - APP  OUT  : [1688140636114] [12778:76445] [EM] >>> [E:38632r S:0 M:185534852] (U) Msg RX from 0:7B06C856CC92BCB5 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
ERROR   15:57:16.118 - APP  OUT  : [1688140636118] [12778:76445] [EM] <<< [E:38632r S:0 M:184811597 (Ack:185534852)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
...
ERROR   15:57:37.998 - PAIR OUT  : [EM] Retransmitting MessageCounter:185534852 on exchange 38632i Send Cnt 2
...
ERROR   15:57:16.118 - APP  OUT  : [1688140636118] [12778:76445] [IN] Received a duplicate message with MessageCounter:185534852 on exchange 38632r
ERROR   15:57:16.118 - APP  OUT  : [1688140636118] [12778:76445] [EM] >>> [E:38632r S:0 M:185534852] (U) Msg RX from 0:7B06C856CC92BCB5 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
ERROR   15:57:16.120 - APP  OUT  : [1688140636120] [12778:76445] [EM] <<< [E:38632r S:0 M:184811598 (Ack:185534852)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
```
and so on until:
```
ERROR   15:57:37.999 - PAIR OUT  : [EM] Failed to Send CHIP MessageCounter:185534852 on exchange 38632i sendCount: 4 max retries: 4
ERROR   15:57:37.999 - PAIR OUT  : [SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
```

So chip-tool sends PBKDFParamRequest, tv-app responds with PBKDFParamResponse, but chip-tool does not get that response.  It keeps retrying the PBKDFParamRequest, tv-app keeps sending acks, chip-tool never gets those, until the whole thing times out.

This is pretty much the same thing as https://github.com/project-chip/connectedhomeip/issues/27524 but with PASE, not CASE....

closed 2023-07-07 by woody-apple