project-chip/connectedhomeip #27584: OperationalState: null CurrentPhase should be permitted on both empty and null PhaseList

robszewczyk, 2023-07-01:
Per latest revisions to the spec

bzbarsky-apple, 2023-08-31:
@robszewczyk What XML changes are needed here?  What spec PR/change is this reflecting?  CurrentPhase and PhaseList are both nullable and have been all along....

robszewczyk, 2023-08-31:
Looking at the underlying spec change: no XML changes are needed. There are potentially changes that should be reflected in the implementation of the OperationalState cluster (e.g. `examples/all-clusters-app/all-clusters-common/src/operational-state-delegate-impl.cpp`), but looking at that implementation: `CurrentPhase` attribute already appears to be settable via SetPhase, and we don't appear to be doing much checking on the arguments to the Phase.