project-chip/connectedhomeip #27598: [1.1] read commissioned fabrics

armysman, 2023-07-03:
### Reproduction steps

1:use our app commissioned matter device.
2:share matter device to google home.
3:read matter device has commissioned by which fabrics.
user OperationalCredentialsCluster->readFabricsAttribute method, but callback just back one which is our app fabrics.use google home can inquired two fabrics.
any another api can replace it or this api has used wrong way?
SDK V1.1

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

82db97695c2c899cc130c2d6a0d72b3c92c144db

### Platform

android

### Type

Common Cluster Logic

### Anything else?

log:
SecureSession[0xb400007bab7c28a8]: Moving from state 'kEstablishing' --> 'kActive'
17:17:18.263 IN                    D  SecureSession[0xb400007bab7c28a8]: Activated - Type:2 LSID:17832
17:17:18.263                       D  New secure session activated for device <0000000000000001, 1>, LSID:17832 PSID:48220!
17:17:18.263 DIS                   D  OperationalSessionSetup[1:0000000000000001]: State change 4 --> 5
17:17:18.264 EM                    D  <<< [E:14449i S:0 M:213532784 (Ack:101962887)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
17:17:18.264 IN                    D  (U) Sending msg 213532784 to IP address 'UDP:192.168.197.112%wlan0:5540'
17:17:18.265 EM                    D  Flushed pending ack for MessageCounter:101962887 on exchange 14449i
17:17:18.269 DMG                   D  SendReadRequest ReadClient[0xb400007adb779dd0]: Sending Read Request
17:17:18.269 EM                    D  <<< [E:14450i S:17832 M:39923516] (S) Msg TX to 1:0000000000000001 [E716] --- Type 0001:02 (IM:ReadRequest)
17:17:18.270 IN                    D  (S) Sending msg 39923516 on secure session with LSID: 17832
17:17:18.270 DMG                   D  MoveToState ReadClient[0xb400007adb779dd0]: Moving to [AwaitingIn]
>>> [E:14450i S:17832 M:114932520 (Ack:39923516)] (S) Msg RX from 1:0000000000000001 [E716] --- Type 0001:05 (IM:ReportData)
17:17:18.388                       D  Found matching exchange: 14450i, Delegate: 0xb400007adb779de0
17:17:18.388                       D  Rxd Ack; Removing MessageCounter:39923516 from Retrans Table on exchange 14450i
17:17:18.388 DMG                   D  ReportDataMessage =
17:17:18.388                       D  {
17:17:18.388                       D  	AttributeReportIBs =
17:17:18.388                       D  	[
17:17:18.388                       D  		AttributeReportIB =
17:17:18.389                       D  		{
17:17:18.389                       D  			AttributeDataIB =
17:17:18.389                       D  			{
17:17:18.389                       D  				DataVersion = 0xc63663b4,
17:17:18.389                       D  				AttributePathIB =
17:17:18.389                       D  				{
17:17:18.389                       D  					Endpoint = 0x0,
17:17:18.389                       D  					Cluster = 0x3e,
17:17:18.389                       D  					Attribute = 0x0000_0001,
17:17:18.389                       D  				}
17:17:18.389                       D  					
17:17:18.389                       D  				Data = [
17:17:18.389                       D  						
17:17:18.389                       D  				],
17:17:18.389                       D  			},
17:17:18.389                       D  			
17:17:18.389                       D  		},
17:17:18.389                       D  		
17:17:18.389                       D  		AttributeReportIB =
17:17:18.389                       D  		{
17:17:18.389                       D  			AttributeDataIB =
17:17:18.389                       D  			{
17:17:18.389                       D  				DataVersion = 0xc63663b4,
17:17:18.389                       D  				AttributePathIB =
17:17:18.389                       D  				{
17:17:18.389                       D  					Endpoint = 0x0,
17:17:18.389                       D  					Cluster = 0x3e,
17:17:18.389                       D  					Attribute = 0x0000_0001,
17:17:18.389                       D  					ListIndex = Null,
17:17:18.389                       D  				}
17:17:18.389                       D  					
17:17:18.389                       D  				Data = 
17:17:18.389                       D  				{
17:17:18.389                       D  					0x1 = [
17:17:18.389                       D  							0x04, 0xc3, 0xca, 0xe1, 0xfe, 0x45, 0x91, 0x54, 0x51, 0x02, 0x90, 0xce, 0x9c, 0x2d, 0x7b, 0x7f, 0x7c, 0x6e, 0x85, 0x23, 0x0f, 0xff, 0xe9, 0x5d, 0x48, 0x3d, 0x33, 0x10, 0xc7, 0x69, 0xda, 0x68, 0x19, 0xcb, 0x3a, 0x9c, 0x74, 0x18, 0x81, 0x58, 0xb9, 0x
17:17:18.389                       D  					] (65 bytes)
17:17:18.389                       D  					0x2 = 4933, 
17:17:18.389                       D  					0x3 = 536872072, 
17:17:18.389                       D  					0x4 = 1, 
17:17:18.389                       D  					0x5 = "" (0 chars), 
17:17:18.389                       D  					0xfe = 2, 
17:17:18.389                       D  				},
17:17:18.389                       D  			},
17:17:18.389                       D  			
17:17:18.389                       D  		},
17:17:18.389                       D  		
17:17:18.389                       D  	],
17:17:18.389                       D  	
17:17:18.389                       D  	SuppressResponse = true, 
17:17:18.389                       D  	InteractionModelRevision = 1
17:17:18.389                       D  }
17:17:18.390 EM                    D  <<< [E:14450i S:17832 M:39923517 (Ack:114932520)] (S) Msg TX to 1:0000000000000001 [E716] --- Type 0000:10 (SecureChannel:StandaloneAck)
17:17:18.391 IN                    D  (S) Sending msg 39923517 on secure session with LSID: 17832
17:17:18.391 EM                    D  Flushed pending ack for MessageCounter:114932520 on exchange 14450i
17:17:20.070 FA                    V  Inactivity, disconnecting from the service

jonsmirl, 2023-07-03:
Check right after you commission your app on the phone and before sharing it to Google Home. There will be two fabrics. Google creates two fabrics on purpose. One is made by the OS and one is made by the your app.  The OS one is made first. It is the one which uses BLE to set the wifi SSID and PW into the device. This has to be done at the OS level since there is no API to retrieve SSID/PW from the OS (security hole).   After the OS commissioning is done the OS does OpenCommissioningWindow and then your app makes a fabric using IP based commissioning. This same process happens on IOS.

bzbarsky-apple, 2023-07-06:
Looks like readFabricsAttribute does a fabric-filtered read and has no way to read "across all fabrics".

chulspro, 2023-08-08:
I would like to share some information that we merged below PR into master. After this PR, I think this issue will be fixed.
- https://github.com/project-chip/connectedhomeip/pull/28208
CC: @andy31415 , @saurabhst

closed 2023-08-09 by andy31415