project-chip/connectedhomeip #27599: [Event-testcase] TC-BINFO-2.2 test Build error & Failure response

manjunath-grl, 2023-07-03:
### Reproduction steps

Summary: 
Scripted BINFO-2.2 test 
1. If we add response for the `StartUp` event `value: { SoftwareVersion: SoftwareVersionValue }` but seeing build error `⛔ TypeError: value.split is not a function` when executed.
2. As per precondition added a subscribeEvent command for `StartUp` event but seeing failure response `Test Failure: zzz_generated/chip-tool/zap-generated/test/Commands.h:33457: CHIP Error 0x0000002F: Invalid argument` when executed.

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Event_Response_Zap_Fail

Reproduction steps:
Step 1:  Add test TC_BINFO_2_2 and run below commands
Step 2: `./scripts/tools/zap_regen_all.py --type tests`
Step 3: ``` scripts/run_in_build_env.sh './scripts/build_python.sh --install_virtual_env out/venv --include_yamltests'
scripts/run_in_python_env.sh out/venv 'pip install -r scripts/setup/requirements.build.txt'
scripts/run_in_python_env.sh out/venv 'pip install -r scripts/setup/requirements.yaml_tests.txt'
                  ./scripts/run_in_build_env.sh \
                     "./scripts/build/build_examples.py \
                        --target linux-x64-chip-tool-no-ble-tsan-clang \
                        --target linux-x64-all-clusters-no-ble-tsan-clang \
                        --target linux-x64-lock-no-ble-tsan-clang \
                        --target linux-x64-ota-provider-no-ble-tsan-clang \
                        --target linux-x64-ota-requestor-no-ble-tsan-clang \
                        --target linux-x64-tv-app-no-ble-tsan-clang \
                        --target linux-x64-bridge-no-ble-tsan-clang \
                        build \
                     "  ```
Step 4: ``` ./scripts/run_in_build_env.sh \
                  "./scripts/tests/run_test_suite.py \
                     --chip-tool -with-python \
                     --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool \
                     run \
                     --iterations 1 \
                     --all-clusters-app ./out/linux-x64-all-clusters-no-ble-tsan-clang/chip-all-clusters-app \
                     --lock-app ./out/linux-x64-lock-no-ble-tsan-clang/chip-lock-app \
                     --ota-provider-app ./out/linux-x64-ota-provider-no-ble-tsan-clang/chip-ota-provider-app \
                     --ota-requestor-app ./out/linux-x64-ota-requestor-no-ble-tsan-clang/chip-ota-requestor-app \
                     --tv-app ./out/linux-x64-tv-app-no-ble-tsan-clang/chip-tv-app \
                     --bridge-app ./out/linux-x64-bridge-no-ble-tsan-clang/chip-bridge-app \
                  " ```

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

7ae5919a046a18c7e60537aff814977ee424b9cf

### Platform

other

### Anything else?

Testplan screenshot:
[image]
Screenshot: 
[image]
[image]

bzbarsky-apple, 2023-07-06:
> TypeError: value.split is not a function

You should disable whatever tests you are using readEvent in when run on darwin-framework-tool, by adding them to the list in examples/darwin-framework-tool/templates/tests/ciTests.json.

Once that's done and codegen has a chance of succeeding, please list which exact command you ran to get your "invalid argument" error (was it step 3 or 4 of your reproduction steps?) and ideally attach the log as a file, not as a screenshot.

manjunath-grl, 2023-07-11:
@bzbarsky-apple I can able to add response after disabling the test in Darwin Json
But for `StartUp` event( step 5: subscribe for StartUp Event) seeing failure response `Test Failure: zzz_generated/chip-tool/zap-generated/test/Commands.h:33457: CHIP Error 0x0000002F: Invalid argument`. Please refer the attached logs.
Execution logs:
[execution_Log.txt](https://github.com/project-chip/connectedhomeip/files/12013574/execution_Log.txt)

bzbarsky-apple, 2023-07-12:
@manjunath-grl OK, so what does the code around line 33457 of zzz_generated/chip-tool/zap-generated/test/Commands.h look like?  Most simply, just attach (not paste) the file here?

manjunath-grl, 2023-07-12:
@bzbarsky-apple Please refer the below diff link added command.h file with the latest code. Now we are seeing error in line `33462`.  `Test Failure: zzz_generated/chip-tool/zap-generated/test/Commands.h:33462: CHIP Error 0x0000002F: Invalid argument`
Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Event_Response_Zap_Fail

Commands.h file:
[Commands.zip](https://github.com/project-chip/connectedhomeip/files/12025087/Commands.zip)

Latest logs:
[Latest_execution_Log.txt](https://github.com/project-chip/connectedhomeip/files/12025071/Latest_execution_Log.txt)

manjunath-grl, 2023-07-12:
From command.h line 33462 I can see that `LogErrorOnFailure(ContinueOnChipMainThread(CHIP_ERROR_INVALID_ARGUMENT));` if fails by default it will throw `INVALID_ARGUMENT` error.
[image]

bzbarsky-apple, 2023-07-12:
Thanks.  So the relevant YAML is:
```
    - label:
          "TH subscribes to StartUp event on the Basic Information cluster of
          the DUT"
      PICS: BINFO.S.E00
      command: "subscribeEvent"
      event: "StartUp"
      minInterval: 30
      maxInterval: 100
```

That should have a section that describes the "response" expected during the priming read for this subscription, which would then get turned into the substep cases in the switch involved.  Right now we are saying we expect no response at all, so we fail when we get a response.

manjunath-grl, 2023-07-13:
Resolved, closing issues.