project-chip/connectedhomeip #27607: [BUG] CHIP Error 0x00000003: Incorrect state when paring device with chip-tool

nmtoan91, 2023-07-04:
### Reproduction steps

I got the error when trying to pair bridged-app with chip-tool.

```
[1688452528.904041][36322:36324] CHIP:IN: (S) Sending msg 50598383 on secure session with LSID: 28015
[1688452528.904083][36322:36324] CHIP:EM: Flushed pending ack for MessageCounter:179198109 on exchange 9690i
[1688452528.904523][36322:36322] CHIP:CTL: Shutting down the commissioner
[1688452528.904548][36322:36322] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1688452528.904604][36322:36322] CHIP:CTL: Shutting down the controller
[1688452528.904620][36322:36322] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1688452528.904633][36322:36322] CHIP:IN: SecureSession[0x7fa74003f630]: MarkForEviction Type:2 LSID:28015
[1688452528.904656][36322:36322] CHIP:SC: SecureSession[0x7fa74003f630, LSID:28015]: State change 'kActive' --> 'kPendingEviction'
[1688452528.904672][36322:36322] CHIP:IN: SecureSession[0x7fa74003f630]: Released - Type:2 LSID:28015
[1688452528.904688][36322:36322] CHIP:FP: Forgetting fabric 0x1
[1688452528.904707][36322:36322] CHIP:TS: Pending Last Known Good Time: 2023-07-04T10:49:47
[1688452528.904872][36322:36322] CHIP:TS: Previous Last Known Good Time: 2023-07-04T10:49:47
[1688452528.904888][36322:36322] CHIP:TS: Reverted Last Known Good Time to previous value
[1688452528.904919][36322:36322] CHIP:CTL: Shutting down the commissioner
[1688452528.904933][36322:36322] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1688452528.904976][36322:36322] CHIP:CTL: Shutting down the controller
[1688452528.904994][36322:36322] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1688452528.905226][36322:36322] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1688452528.905270][36322:36322] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1688452528.905299][36322:36322] CHIP:FP: Shutting down FabricTable
[1688452528.905322][36322:36322] CHIP:TS: Pending Last Known Good Time: 2023-07-04T10:49:47
[1688452528.905468][36322:36322] CHIP:TS: Previous Last Known Good Time: 2023-07-04T10:49:47
[1688452528.905484][36322:36322] CHIP:TS: Reverted Last Known Good Time to previous value
[1688452528.905665][36322:36322] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-uEXqVI)
[1688452528.906021][36322:36322] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1688452528.906067][36322:36322] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1688452528.906085][36322:36322] CHIP:DL: Inet Layer shutdown
[1688452528.906102][36322:36322] CHIP:DL: BLE shutdown
[1688452528.906130][36322:36322] CHIP:BLE: CancelBleIncompleteConnection() failed, err = src/ble/BleLayer.cpp:372: CHIP Error 0x00000003: Incorrect state
[1688452528.906151][36322:36322] CHIP:DL: System Layer shutdown
```

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

bzbarsky-apple, 2023-07-06:
This looks like fallout from #27304: we are trying to cancel things when we are not even initialized....

@cuizelin99 can you take a look please?

bzbarsky-apple, 2023-07-24:
Fixed by https://github.com/project-chip/connectedhomeip/pull/28143

closed 2023-07-24 by bzbarsky-apple