project-chip/connectedhomeip #27621: [BUG] wrong value returned when calling values which are long types on Android 32-Bit OS (armeabi-v7a)

buddman1208, 2023-07-05:
### Reproduction steps

**Tested Environment: Android Device** 
- Chromecast 4K Android 12 (armeabi-v7a)
- GiGA Genie 3 (KAON KI1100) Android 11 (armeabi-v7a)

**ABI: armeabi-v7a (32bit)** 

1. Problem
I am trying to get end-device's specific value, such as deviceTypeId of matter light bulb, so i am calling code below: 
```
 @OptIn(ExperimentalCoroutinesApi::class)
    suspend fun loadDeviceType(): Long {
        val descriptorCluster = getDescriptorCluster(context, deviceId)

        return suspendCancellableCoroutine { continuation ->
            descriptorCluster.readDeviceTypeListAttribute(object: DeviceTypeListAttributeCallback{
                override fun onSuccess(valueList: MutableList<ChipStructs.DescriptorClusterDeviceTypeStruct>?) {
                    continuation.resume(valueList!![0].deviceType) {}
                }

                override fun onError(ex: Exception?) {
                    continuation.resumeWithException(ex ?: Exception("Unknown error"))
                }
            })

            continuation.invokeOnCancellation {
                Timber.tag(TAG).e("loadDeviceType cancel, $it")
            }
        }
    }
```

and returned value was : 
on arm64-v8a (Google Pixel 6, Samsung Galaxy S20 FE, Samsung Note 10, etc..)
```
269
```
on armeabi-v7a (Chromecast 4K, GiGA Genie 3 (KAON KI1100)
```
-5178014922610311676
```

**As you can see, returned deviceTypeId is different between architectures. (32bit <-> 64bit)**
Returned value from 32bit os is incorrect.

So I looked at logcat on android, and i found out that deviceTypeId value was returned correctly inside sdk. (both 32bit, 64bit)
Look at my logcat below: these are appeared logs when i called descriptorCluster.readDeviceTypeListAttribute().

```
ReportDataMessage =
{
	AttributeReportIBs =
	[
		AttributeReportIB =
		{
			AttributeDataIB =
			{
				DataVersion = 0x82102a0d,
				AttributePathIB =
				{
					Endpoint = 0x1,
					Cluster = 0x1d,
					Attribute = 0x0000_0000,
				}
					
				Data = [
						
				],
			},
			
		},
		
		AttributeReportIB =
		{
			AttributeDataIB =
			{
				DataVersion = 0x82102a0d,
				AttributePathIB =
				{
					Endpoint = 0x1,
					Cluster = 0x1d,
					Attribute = 0x0000_0000,
					ListIndex = Null,
				}
					
				Data = 
				{
					0x0 = 269, 
					0x1 = 1, 
				},
			},
			
		},
		
	],
	
	SuppressResponse = true, 
	InteractionModelRevision = 1
}
```
As far as I can predict, the value is normally obtained inside the sdk, 
but when the value is returned through JNI on 32bit android os, the wrong value appears.

This error is not just for deviceTypeId, but every (that i checked) value which type is long.

Thank you for assistance.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

17/05/23 / v1.1-branch / 82db97695c2c899cc130c2d6a0d72b3c92c144db

### Platform

android

### Platform Version(s)

tested on android 11, 12

buddman1208, 2023-07-05:
Additional: 

These are also affected to getting device cluster list. Have a look:

32bit (armeabi-v7a)
```
clusterList=[ClusterList(clusterDecimal=-5517825536783761780), ClusterList(clusterDecimal=-5517825536783761780), ClusterList(clusterDecimal=-5517825536783761780), ClusterList(clusterDecimal=-5517825536783761780), ClusterList(clusterDecimal=-5517825536783761780), ClusterList(clusterDecimal=-5517825536783761780)])]
```

64bit (arm64-v8a)
```
clusterList=[ClusterList(clusterDecimal=3), ClusterList(clusterDecimal=4), ClusterList(clusterDecimal=6), ClusterList(clusterDecimal=8), ClusterList(clusterDecimal=29), ClusterList(clusterDecimal=768)])]
```

buddman1208, 2023-07-05:
If you need other information, I will deliver it as quick as possible. Thanks

bzbarsky-apple, 2023-07-07:
@andy31415 @yunhanw-google

yunhanw-google, 2023-07-27:
@buddman1208 please consider to use generic IM API for cluster interaction, https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L837
We have decoupled generated chip cluster API in java/jni side, and plan to stop the maintenance on that part soon. We have migrate android chip-tool to use the above generic IM API. You can see the example code from android chip-tool

If this issue is still reproducible, please let me know, thanks

buddman1208, 2023-07-27:
@yunhanw-google Thanks, We'll give it a try and let you know whether it works well or not.

joonhaengHeo, 2023-07-31:
I checked this issue, and it seems to be an issue that appears during the static_cast process.
https://github.com/project-chip/connectedhomeip/tree/master/src/controller/java/zap-generated

For testing, I changed the file created by ZAP(CHIPReadCallbacks.cpp) as follows, and it is confirmed that normal values are transmitted.

As-Is :
` auto & entry_0 = iter_arrayListObj_0.GetValue();`
`         jobject newElement_0;`
`         jobject newElement_0_deviceType;`
`         std::string newElement_0_deviceTypeClassName = "java/lang/Long";`
`         std::string newElement_0_deviceTypeCtorSignature = "(J)V";`
`         chip::JniReferences::GetInstance().CreateBoxedObject<uint32_t>(newElement_0_deviceTypeClassName.c_str(),`
`                                                                        newElement_0_deviceTypeCtorSignature.c_str(),`
`                                                                        entry_0.deviceType, newElement_0_deviceType);`

To-Be : 
` auto & entry_0 = iter_arrayListObj_0.GetValue();`
`         jobject newElement_0;`
`         jobject newElement_0_deviceType;`
`         std::string newElement_0_deviceTypeClassName = "java/lang/Long";`
`         std::string newElement_0_deviceTypeCtorSignature = "(J)V";`
**`         jlong jniDeviceType = static_cast<jlong>(entry_0.deviceType);`**
`         chip::JniReferences::GetInstance().CreateBoxedObject<jlong>(newElement_0_deviceTypeClassName.c_str(),`
`                                                                        newElement_0_deviceTypeCtorSignature.c_str(),`
`                                                                        jniDeviceType, newElement_0_deviceType);`

yunhanw-google, 2023-08-01:
agree, let's put the fix on it, I think previously i fix most the static_cast, and forgot the update it in zap file.

joonhaengHeo, 2023-08-02:
@bzbarsky-apple @yunhanw-google 
I uploaded the fixed code.

However, the ZAP creation script has some problems, so this script also needs to be modified.
Could you please review if this fix is in the right direction?