project-chip/connectedhomeip #27631: [BUG] OpenThread and number of multicast addresses (`OPENTHREAD_CONFIG_IP6_MAX_EXT_MCAST_ADDRS`)

dtodor, 2023-07-05:
### Reproduction steps

What is the minimal/expected number of multicast addresses for Thread devices? We are testing group communication with an EFR32 board, and we cannot use more then 2 groups, which I guess results from this definition:

https://github.com/openthread/openthread/blob/6736ef267673d1e15d528067c3433377597a02b2/src/core/config/ip6.h#L57

When the device is started with more then 2 groups set up, we can see the following output:
```
[00:00:00.738][info  ][SVR] Joining Multicast groups
[00:00:00.740][info  ][IN] Joining Multicast Group with address UDP:[ff35:40:fd00:0:0:0:100:7d0]:5540
[00:00:00.742][info  ][IN] Joining Multicast Group with address UDP:[ff35:40:fd00:0:0:0:100:7d1]:5540
[00:00:00.745][info  ][IN] Joining Multicast Group with address UDP:[ff35:40:fd00:0:0:0:100:bb8]:5540
[00:00:00.745][error ][SVR] Error when trying to join Group 3000 of fabric index 1 : 4000003
```

### Bug prevalence

Always reproducible

### GitHub hash of the SDK that was being used

e278aa42166d29ad49156d549376a96d0a3c82a0

### Platform

efr32

cjandhyala, 2023-07-05:
@dtodor The MaxGroupsPerFabric attribute in the Group Key Management Cluster will have the number that maximum number of groups that the node supports per fabric

dtodor, 2023-07-06:
The EFR32 device reports the following values:
```
OperationalCredentials::SupportedFabrics = 5
GroupKeyManagement::MaxGroupsPerFabric = 4
```

So the value of `OPENTHREAD_CONFIG_IP6_MAX_EXT_MCAST_ADDRS` should be 20, right? Currently there is a discrepancy - we can add and configure more than 2 groups, but joining more than 2 multicast groups fails.

dtodor, 2023-07-06:
BTW, what about changing the name of the `MaxGroupKeysPerFabric` attribute to `MaxGroupKeySetsPerFabric`:

```
11.2.7.4. MaxGroupKeysPerFabric Attribute
This attribute SHALL indicate the maximum number of group key sets this node supports per fab­ ric. The value of this attribute SHALL be set according to the minimum number of group key sets to support as specified in Group Limits.
```

bzbarsky-apple, 2023-07-06:
> BTW, what about changing the name of the MaxGroupKeysPerFabric attribute to MaxGroupKeysetsPerFabric:

This would have been a reasonable suggestion a year ago, but at this point there's a bunch of shipping API with the existing name.

Past that, the spec says:

> A node SHALL support one IPv6 multicast group membership for every operational group it supports.

so if something has 5 supported fabrics and 4 groups per fabric, but only allows 2 multicast addresses, then it's very much non-conformant.

@Damian-Nordic should we be setting `OPENTHREAD_CONFIG_IP6_MAX_EXT_MCAST_ADDRS` as needed when we compile openthread?

Damian-Nordic, 2023-07-06:
@bzbarsky-apple Yes, that seems right that `OPENTHREAD_CONFIG_IP6_MAX_EXT_MCAST_ADDRS` should be increased.

bzbarsky-apple, 2023-07-08:
I guess my question was whether we can have it automatically set based on the other two constants.

Damian-Nordic, 2023-07-08:
Oh, then the problem might be that most device manufacturers don't build OpenThread from sources but use OpenThread libraries built and certified by platform vendors (to take advantage of certification by inheritance). So I guess platform vendors must simply ensure that they configure a sane value for `OPENTHREAD_CONFIG_IP6_MAX_EXT_MCAST_ADDRS` in the libraries they provide.

dtodor, 2023-07-09:
I think it would be important to have the Matter SDK examples correctly configured and running according to the specification. It might very well be the case that the device manufacturers are not aware of this restriction. I will perform some tests with certified devices and report back.

dtodor, 2023-07-12:
I just did some tests with a Matter certified device. The device reports:
```
GroupKeyManagement::MaxGroupsPerFabric: 3
GroupKeyManagement::MaxGroupKeysPerFabric: 3
AccessControl::AccessControlEntriesPerFabric: 3
AccessControl::SubjectsPerAccessControlEntry: 4
```
I am able to add more than 3 groups without any errors. The device is controllable only with the first group. I think this is a Nordic device, not sure though:
```
BasicInformation::DataModelRevision: 1
BasicInformation::VendorName: Eve Systems
BasicInformation::ProductName: Eve Energy 20EBO8301
BasicInformation::HardwareVersion: 1
BasicInformation::SoftwareVersion: 6200
BasicInformation::SoftwareVersionString: 3.1.0
```

dtodor, 2023-07-12:
I just saw that there is a newer software version (3.2.1) for this device, will update it and redo the tests.

damien0x0023, 2024-10-24:
@dtodor 
We encountered a similar issue with our OpenThread device, where it could only respond to the first group.

Based on the discussion in this thread, we increased the values for `OPENTHREAD_CONFIG_IP6_MAX_EXT_MCAST_ADDRS` and `CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT`. 
After making these changes, it was able to be controlled by multiple groups.

I hope our experience can help you resolve this issue as well.

NateBowen, 2024-10-28:
@damien0x0023 - Which device platform are you using? ESP32, Nordic, Linux?

JoshMJohnston, 2024-10-30:
> Oh, then the problem might be that most device manufacturers don't build OpenThread from sources but use OpenThread libraries built and certified by platform vendors (to take advantage of [...]

On this note, those that are using Nordic devices, make sure you set `CONFIG_OPENTHREAD_SOURCES=y` in order for any of your OpenThread related KCONFIG values to actually take effect. See ticket [here](https://devzone.nordicsemi.com/f/nordic-q-a/115851/matter-cannot-add-an-endpoint-to-more-than-two-groups).

damien0x0023, 2024-12-25:
> @damien0x0023 - Which device platform are you using? ESP32, Nordic, Linux?

Telink B9x, W91 and TLx platforms.
Here is our PR #36290.