project-chip/connectedhomeip #27632: [Tooling] BLE Cancel Connection

jjm202, 2023-07-05:
### Reproduction steps / Feature

Build chip-tool using a linux/arm64 based Docker container. After installing the necessary build tools and sysroot:

- export SYSROOT_AARCH64=/opt/ubuntu-22.04.1-aarch64-sysroot	
- checkout source code
- ./scripts/checkout_submodules.py --shallow --platform **linux**
- source ./scripts/bootstrap.sh
- ./scripts/gn_build_example.sh examples/chip-tool out

chip-tool builds then is copied to the R Pi and used normally. The question is why does the chip-tool that is packed with the test harness image not produce the following errors (these appear to be benign, but still inconsistent). 

```
[1688578543.970738][6727:6729] CHIP:BLE: CancelBleIncompleteConnection() failed, err = src/platform/Linux/BLEManagerImpl.cpp:732: CHIP Error 0x0000002D: Not Implemented
.
.
.
 CHIP:BLE: CancelBleIncompleteConnection() failed, err = src/ble/BleLayer.cpp:372: CHIP Error 0x00000003: Incorrect state
```

### Platform

raspi

### Type

Manually tested with SDK

bzbarsky-apple, 2023-07-06:
> The question is why does the chip-tool that is packed with the test harness image not produce the following errors

Probably because it's built from a version that predates https://github.com/project-chip/connectedhomeip/pull/27304 ?

bzbarsky-apple, 2023-07-06:
Duplicate of #27607

closed 2023-07-06 by bzbarsky-apple